sui77/rc-switch #318: Libary example stays quiet bit reciever is bitbanging.

Stephan3, 2020-01-09:
Hello!

I tested the ReceiveDemo advanced and simple. Both staying silent on serial.

mySwitch.available()

never triggers. I use theese receivers https://www.amazon.de/gp/product/B071J2Z3YK/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

If i connect a Logic analyzer to the receiver module its bitbanging. If i push a button on the remote i try to capture the bitbanging sequences are changing. After releasing a button all is 1 for a while.
Is there a special Setup or receiver i have to use with this library?

Thanks 
Stephan

1technophile, 2020-01-09:
Hi,

these receivers are OK, but did you connected them to a PIN with an interrupt ?

Stephan3, 2020-01-09:
Thanks for your quick reply. I think i did it right, Here are some details:

Its a Arduino nano . The receiver DATA is connected to D2 of Nano. 5V and GND is connected to Arduino 5V and GND. Nano is supplied by usb of a laptop,

With this sketch i can see that the device is bitbanging. If i push a button on the remote i can see a lot of data passing by.

```C
#include "RCSwitch.h"
#include <stdlib.h>
#include <stdio.h>
RCSwitch mySwitch = RCSwitch();

int j = 0;

void setup() {
 Serial.begin(115200);
 mySwitch.enableReceive(0);
 Serial.println( "bin da" );
}

void loop() {

 j=j+1;

 Serial.print( digitalRead(2) );

 if(j==31){
  Serial.println("");
  j=0;
 }
}
```

If i push no button, there seems to be a lot of noise. The remote is the following.

https://www.jalousiescout.de/Funktechnik-/-Funksteuerung/Auswahl-nach-Hersteller/Jarolift/Jarolift-Funkwandsender-Timer-4-Kanal-TDRCT-04W.html?gclid=CjwKCAiAu9vwBRAEEiwAzvjq-8xxRRxmkeXgqW3PDcI_NeJmBlZwznAOiiHEJlN7_hctUDLmnIJM1RoCRmAQAvD_BwE

I really apreciate your help. My plan is to sniff each button and send it with a transmitter on a esp32 using alexa.

1technophile, 2020-01-11:
>  If i push a button on the remote i can see a lot of data passing by.

An hypotesis could be that your remote is not recognised by rcswitch.

To verify that your setup is working you could try to set another uC with an emitter, load into it the senddemo example and see if you receive something with the receiver one.

At final could you detail what you would like to achieve ?

Stephan3, 2020-01-11:
Thanks meanwhile i realized that the device is using Keeloq. Therefore rc-switch is not the right tool. (correct me if i am wrong)

1technophile, 2020-01-11:
Maybe try this version that I have not yet merged:
https://github.com/1technophile/rc-switch