sui77/rc-switch #320: D1 Mini ESP-8266

aramaziakin, 2020-01-29:
I'm trying to get the receive demo working on a D1 Mini but as far as I can tell it crashes everytime I start it with this being printed to the serial monitor several times a second.
```
User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: cont
sp: 3ffffed0 end: 3fffffc0 offset: 0000
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  000000fe 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00ff0000  
3fffff00:  5ffffe00 5ffffe00 00000020 00000000  
3fffff10:  00000003 00000000 3ffee48c 402021ba  
3fffff20:  40100402 3ffef14c 3ffef134 402021d0  
3fffff30:  0000001c 0001c200 3ffee48c 402026e1  
3fffff40:  00000000 40203195 ffffffff 00000001  
3fffff50:  feefeffe 00000001 3ffee464 3ffee4cc  
3fffff60:  3fffdad0 00000000 3ffee48c 40202790  
3fffff70:  00000000 feefeffe feefeffe 4020113c  
3fffff80:  0001c200 0000001c 00000000 40201150  
3fffff90:  3fffdad0 00000000 3ffee48c 40201040  
3fffffa0:  feefeffe feefeffe feefeffe 40201e10  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100d0d  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
ISR not in IRAM!
```
Attaching a 433 receiver board doesn't make any difference, neither does changing the receiver pin in the code, and the board works fine with the Wifi scan example.

aramaziakin, 2020-01-29:
I just remember I have another ESP8266 board lying around (a NodeMCU) but I'm getting exactly the same error with that too.

1technophile, 2020-02-02:
Did you try with the last version of arduino-esp8266 platform?

1technophile, 2020-02-04:
Could you indicate the version of arduino-esp8266 used in board manager ?

Martin-Laclaustra, 2020-02-07:
This line could be the culprit:
https://github.com/sui77/rc-switch/blob/a847651c3238856611fa3b9b6a48eb1f87c0313c/RCSwitch.cpp#L601

According to https://github.com/sui77/rc-switch/issues/46#issuecomment-559992175 it could be replaced with:
```
if ((A - B)>0) return (A - B); else return -(A - B) ;
```
This could be an error occurring randomly depending on the rest of the sketch (the compiler decides what to left out of IRAM).
@aramaziakin , @Lazare-C . Please try it and see if it solves the problem. Although I could not reproduce it, if it solves your problem it would be worth to be changed in the library.