sui77/rc-switch #324: Doesn't work with TTGO T-Call  (ESP32)

Terxeno, 2020-02-08:
Hi!
I try to use ESPiLite c TTGO T-Call (ESP32) and srx882, but unfortunately nothing works, (RECEIVER) there is no answer at all. Although the same code works on The Wemos D1 mini, Arduino.
Tried different pins, different code, but all the same does not work.

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitch.enableReceive(13);  // Receiver on interrupt 0 => that is pin #2
}

void loop() {
  if (mySwitch.available()) {
    output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
    mySwitch.resetAvailable();
  }
}
```

1technophile, 2020-02-08:
Hi,
On ESP32 I use to connect the receiver on pin 27, could you try with it?

Terxeno, 2020-02-08:
Can't use 27 pin, it's sim800 pin connect (TTGO T-Call board ESP32+SIM800L)
#define MODEM_TX             27

1technophile, 2020-02-09:
> Do you connect CS on srx882 to +5v?
No need to connect CS to a pin

> Can't use 27 pin, it's sim800 pin connect (TTGO T-Call board ESP32+SIM800L)
#define MODEM_TX 27

>The only ones I'm sure about are 26 and 27 that are used by the LORA module.

Terxeno, 2020-02-09:
I tried using 27 pins, but then the controller just doesn't start.

1technophile, 2020-02-12:
One user of OpenMQTTGateway report that esp32 is working with an RF receiver on pin 17 and an RF emitter on pin 12 for your info

https://community.openmqttgateway.com/t/433mhz-lora-with-cheap-sensors-outlets/339/8?u=1technophile

Terxeno, 2020-02-13:
Unfortunately, there is no such pin, on this board, or rather it is signed as SD2, but nothing can be connected to it.
But I've found where to connect! You always have to read the documentation!

And it says:

> GPIO4 is the default output GPIO of the 1 Hz pulse generator.
> **GPIO18 is the default pulse input GPIO**. We need to short GPIO4 and GPIO18.
> GPIO5 is the default control signal, which can be left floating with internal pull up, or connected to Ground (If GPIO5 is left floating, the value of counter increases with the rising edges of the PWM pulses. If GPIO15 is connected to Ground, the value decreases).

So, after connecting DATA to pin 18, everything worked!

1technophile, 2020-02-16:
Nice, thanks for the info

closed 2020-03-09 by 1technophile