sui77/rc-switch #329: Decoding, no output in serial monitor

dakoeller, 2020-03-10:
Dear All,

I am trying to decode the below 433mhz rf switch with ev1527, but without any luck. The serial monitor stays blank. I use a nodemcu  and a 5v rf receiver as a hardware. The nodemcu is powered via usb, the receiver is powered from the nodemcu vin pin. I used the receive_demo_advanced code and also tried the simple one but no success. Anybody have a clue what is the issue?

Thanks, Dave

[image]
[image]

NorthernMan54, 2020-03-10:
First comment your missing an Antenna, needs to have attached by the coil

Second comment, I’m in the same boat as your self but I’m running with a scope ( no software yet on the esp8266) attached to the output pins and see nothing.  I figure I have a dud board.

NorthernMan54, 2020-03-10:
Also does your nodemcu supply 5 volts on the vin pin? I have one type of nodemcu that doesn’t.

dakoeller, 2020-03-10:
Thank you for your comment! I tried now with a wire attached as an antenna but still nothing. Yes the board supplies 5v on the Vin I verified it with a multimeter. It is really annoying, it should work, the switch is supported by the rcswitch folder.

dakoeller, 2020-03-10:
But often catches some random noise which is not the my switch:
[image]

1technophile, 2020-03-10:
Hello, did you tried with another pin ?
Here is a list of the compatible ones:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1864843263

dakoeller, 2020-03-10:
Hello there! Thank you! I tried all the pins according to the link but none of it works. Still nothing happens in the serial monitor when i press the switch.

1technophile, 2020-03-10:
> I tried now with a wire attached as an antenna but still nothing

Does it measure 17.3 cm ?

> Hello there! Thank you! I tried all the pins according to the link but none of it works. Still nothing happens in the serial monitor when i press the switch.

Do you have another nodemcu with an emitter to load into it a senddemo and see if you have a compatibility issue with your switch?

dakoeller, 2020-03-10:
Yes the antenna is ~173mm. Unfortunatelly I only have receivers, but I hoped it would work, because ev1527 is among the compatible devices. I also found several tutorials using the same hardwares.

dakoeller, 2020-03-26:
Hi All! I made more testing, as 1technophile suggested I used an emitter to send codes via senddemo. This way there were no problem, the receiver put the codes to the serial monitor. So indeed the switch is the issue. I also found a schematic for it, and Ive found one interesting stuff: this one uses a ev1527 encoder chip, which has an output pin txd on the picture. But the output is not connected to anywhere. As for my understanding 433mhz transmitters should work the way they send the precoded key through the tx pin to the transmitter unit. Can someone explain how this switch works then??

[image]