sui77/rc-switch #330: Help with adding new Remote

Delta-38, 2020-03-30:
Hi, I've been using this library for a little while and really like it, and I am trying to use to control four FAAC brand driveway gates.
I only have copies of copies of the original remotes, as the system is at least 20 years old. 
I think I've done my homework with the Wiki and Add New Remote sections. 
I used the protocollessreceiver branch by Martin Laclaustra to analyze the protocol and confirm the timings I had found: 
`
Cancello Uno 
Received 2080 / 12bit Protocol: 0
data bits of pulse train duration: 12416
proposed protocol: { 345, { 35, 2 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
12228 648 
608 376 292 748 292 740 304 732 308 732 308 736 600 440 316 724 
308 732 312 724 316 728 316 724 

Cancello Due
====
Received 2095 / 12bit Protocol: 0
data bits of pulse train duration: 10168
proposed protocol: { 282, { 37, 2 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
10428 596 
576 280 344 532 184 676 240 604 204 636 260 580 492 360 228 660 
480 300 520 316 556 296 
====

Cancello Tre
====
Received 47 / 12bit Protocol: 6
data bits of pulse train duration: 12472
proposed protocol: { 346, { 35, 2 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
12124 652 
256 780 256 784 256 776 272 772 264 780 264 768 600 432 280 760 
604 448 600 432 612 436 604 436 
====
Cancello Quattro
====
Received 896 / 12bit Protocol: 0
data bits of pulse train duration: 12092
proposed protocol: { 336, { 35, 2 }, { 1, 2 }, { 2, 1 }, true }
====
first level down
11904 652 
284 724 244 768 568 444 560 436 568 440 240 772 236 772 236 764 
244 764 244 768 240 764 244 768 
====`

I noticed that among repetitions there was a little drift in timing (4-12 microseconds but it seemed more due to the remote being used than to arduino).

I added these two different protocols to a fork of rc-switch, and tried ReceiceDemo, which would now see the same output from SendDemo (running on a different arduino) and the remotes. 

However, even attaching a 17.3cm antenna to the module (Range Tested with an RXB6 receiver to 10meters in house through walls) and walking straight up to the control box for the individual gates, none of them open. 

I have also looked at the plots with the raw data on https://test.sui.li/oszi/
And the plots look to have the same sequence and spacing.

The transmission code I used in senddemo was like this (of course replicated for the other gates as well):
`
#define G134Protocol 7
#define G2Protocol 8
int repetitions = 100;
void sendGateOneTrain(){
  mySwitch.setProtocol(G134Protocol);
  mySwitch.setPulseLength(345);
  mySwitch.setRepeatTransmit(repetitions);
  mySwitch.send(2080,12);
}`

and similar for the other gates. 

The receiver shows the same identical data, but the gates don't open. 
Could you point me in the right direction? 

Hardware: 
Duplicated Remote: Remocon LRT-1 / Remocon Y2K10L
Board: Arduino Uno
Test Receiver: RXB6
Test Transmitter: XKFST

Thanks in advance and willing to buy you a coffee :)
[image]
[image]
[image]
[image]

Delta-38, 2020-03-30:
Don't know why the text shifted size...

typh289, 2020-05-17:
With a FAAC gate with very similar readings i had success using the protocol:  
 { 330, { 34,  1 }, {  1,  2 }, {  2,  1 }, true }
Found in another older issue, and there are good resources there that might help you as they did with me :)
https://github.com/sui77/rc-switch/issues/135