sui77/rc-switch #332: Receive decodes the signal correctly but transmits different data

olonsoft, 2020-04-10:
I have a remote control that controls a device.
When I click a button on the remote, the device responds to the command.
I decode the signal with rc-switch (ReceiveDemo_advanced.ino) which is the following:
```
Decimal: 16380978 (24Bit) Binary: 111110011111010000110010 Tri-State: not applicable PulseLength: 294 microseconds Protocol: 1
Raw data: 9185,875,314,873,317,862,326,856,332,852,336,260,918,264,918,858,332,851,336,850,340,846,336,849,340,249,931,847,342,251,932,255,924,260,925,261,921,856,334,848,338,256,926,261,923,852,333,260,925,

Decimal: 16380978 (24Bit) Binary: 111110011111010000110010 Tri-State: not applicable PulseLength: 296 microseconds Protocol: 1
Raw data: 9189,878,314,867,322,862,327,859,330,853,335,258,924,262,922,854,334,850,336,850,341,843,344,843,345,249,934,843,346,248,932,254,930,257,925,262,925,852,335,849,339,256,924,262,924,850,336,256,929,

Decimal: 16380978 (24Bit) Binary: 111110011111010000110010 Tri-State: not applicable PulseLength: 296 microseconds Protocol: 1   
Raw data: 9195,878,312,871,320,862,325,856,334,853,334,257,927,262,920,854,334,851,337,849,342,843,344,843,345,246,937,844,343,252,932,256,929,257,926,258,925,852,335,852,340,253,926,261,924,856,331,261,921,
```
_(Note 1. that the rc-switch found the 3 transmitions while my remote sends 7).
Note 2. the remote uses EV1527 encoding)_
I visualize this signal at https://test.sui.li/oszi/ which is this:
[visualized signal](https://imgur.com/FV6FlMB)
I also record the transmission with my SDR and display it on Audacity. This signal is the bottom of the following graph.  
I then transmit the signal (with SendDemo.ino) but my device does not respond. I record the rc-switch transmitted signal again with SDR and I see that the signal transmitted is completely different from the one I first decoded. (look the top wave on the following graph):
[Audacity recorded waveforms](https://imgur.com/3NSyu2W)
I am using esp8266 wemos D1. The transmitter and receiver are: [those](https://www.banggood.com/433MHz-100M-Wireless-Transceiver-Module-Kit-Transmitter-Receiver-2pcs-Copper-Spring-Antenna-p-1179985.html)
Can someone explain what is wrong?

Martin-Laclaustra, 2020-04-22:
Use SimpleRCScanner with your original remote and your signal transmitted with the library.
Post the raw values.
Read the wiki on new remotes.

olonsoft, 2020-04-23:
I did some tests with SimpleRcScanner on an arduino and I post the results:
My remote control sends (received from SimpleRcScanner on Arduino):
```
0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,60172,304,9148,880,308,876,316,860,328,856,332,852,336,256,928,256,924,852,340,844,344,844,344,840,348,836,352,240,940,840,352,240,940,248,936,248,932,256,932,844,344,840,348,248,936,248,932,848,344,248,936,252,9196,868,324,860,328,860,328,848,344,848,340,252,932,252,932,844,348,840,348,836,348,840,352,832,360,236,944,836,352,240,944,244,940,248,936,248,932,848,344,840,348,248,936,252,936,836,348,248,936,252,9200,872,320,860,328,852,340,848,340,848,344,248,936,248,936,844,344,840,348,836,356,828,360,828,360,236,948,828,364,232,944,244,944,244,936,248,940,836,352,836,352,244,940,244,936,848,344,248,936,248,9208,868,324,856,332,852,340,844,344,840,348,248,936,252,932,844,344,840,352,836,352,832,356,832,360,236,944,836,352,240,944,244,940,244,936,256,932,844,348,836,352,244,940,248,932,844,348,248,936,248,9208,868,324,856,332,852,340,844,348,840,348,240,944,248,932,844,348,840,348,836,356,828,360,828,360,236,948,832,352,240,948,244,940,244,940,248,936,844,344,840,352,240,940,248,936,844,344,248,944,244,9208,868,324,860,332,852,336,848,340,844,348,244,940,248,936,844,344,844,348,836,352,832,356,832,356,240,948,828,360,236,944,240,948,240,944,248,932,844,344,844,352,240,940,248,936,840,352,248,936,248,9212,864,324,860,328,856,336,848,344,840,348,244,940,248,936,844,348,840,348,836,352,836,356,828,360,236,948,832,360,232,948,244,940,244,940,252,932,844,344,840,348,248,940,244,940,844,344,248,936,252,9208,872,320,860,328,860,332,852,336,848,344,248,936,256,928,848,340,848,344,840,348,836,356,832,360,236,940,840,352,240,944,244,940,248,936,248,944,664,
```
RC-Switch (ReceiveDemo_advanced.ino on a ESP8266) decodes:
```
Decimal: 16380978 (24Bit) Binary: 111110011111010000110010 Tri-State: not applicable PulseLength: 296 microseconds Protocol: 1   
Raw data: 9195,878,312,871,320,862,325,856,334,853,334,257,927,262,920,854,334,851,337,849,342,843,344,843,345,246,937,844,343,252,932,256,929,257,926,258,925,852,335,852,340,253,926,261,924,856,331,261,921,
```
RC-Switch sends (above decoded data with SendDemo.ino on ESP8266, received with SimpleRcScanner on Arduino):
```
mySwitch.setProtocol(1);
mySwitch.setPulseLength(296);
mySwitch.setRepeatTransmit(7);
mySwitch.send("111110011111010000110010");
0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,12,40008,804,408,780,416,764,448,736,452,732,460,140,1052,136,1040,736,452,736,452,732,468,720,480,704,476,124,1068,712,468,128,1068,120,1052,140,1044,144,1060,720,460,724,472,124,1048,140,1044,736,472,124,1068,124,9304,768,424,760,440,744,452,732,472,716,476,120,1088,100,1052,728,460,728,480,704,476,712,480,704,472,128,1064,712,472,128,1060,128,1056,136,1044,144,1044,732,460,728,468,128,1048,144,1044,732,468,128,1052,136,9320,756,432,752,436,748,464,724,464,720,468,128,1052,140,1048,728,460,728,468,716,496,692,484,704,488,108,1076,704,468,128,1060,128,1052,140,1032,156,1040,736,456,732,460,136,1060,132,1044,732,456,140,1052,140,9308,764,428,760,432,752,456,728,452,732,472,124,1064,128,1040,736,456,732,468,716,476,712,472,716,480,116,1064,716,484,112,1060,132,1056,132,1052,140,1036,744,456,724,472,124,1064,128,1044,732,468,132,1044,144,9296,780,416,768,440,744,460,724,460,728,464,132,1060,128,1040,740,480,704,468,720,464,720,480,708,484,112,1072,708,496,100,1068,124,1060,128,1056,136,1036,740,460,724,484,112,1064,128,1060,720,452,144,1060,128,9300,776,420,764,448,736,448,736,452,736,480,116,1052,136,1044,736,472,712,472,716,476,712,480,704,496,100,1064,716,476,120,1072,120,1060,128,1064,128,1044,732,492,696,472,124,1064,124,1040,740,456,140,1044,144,9308,768,432,752,440,744,448,736,468,720,472,128,1060,124,1048,732,456,728,464,724,460,724,488,704,476,116,1068,712,496,100,1072,120,1048,144,1052,136,1040,736,480,708,476,120,1048,140,1036,744,468,128,1044,144,
```

Martin-Laclaustra, 2020-04-23:
Try this one:
{ 400, { 1, 23 }, { 1, 2 }, { 2, 1 }, false }
Your overall timings are accurate (compare your readings entering them one after another in https://test.sui.li/oszi/ ) but:
Your transmitter seems slow or weak (high pulses are shorter that they should, for both, short and long pulses).
So the above numbers are a "tweaked" protocol, to ensure that short high pulses are longer (hoping that long high pulses do not shorten too much).
Record your signal again to compare. If your receiver still does not recognize your signal, try feeding 5V to your transmitter (if compatible. Be careful with your microcontroller - try the arduino).
It seems to be a hardware problem. You might need a different transmitter.

ninora, 2020-04-24:
@Martin-Laclaustra,
Please check your @github.com email account for a message regarding this library. 
Thanks

Martin-Laclaustra, 2020-04-24:
@ninora 
I am not the maintainer of this library.
Please explain why you need to reach me and provide your contact details.

olonsoft, 2020-04-24:
@Martin-Laclaustra much better:
original remote:
```
936,252,936,256,932,260,924,852,340,852,340,252,932,260,924,856,336,256,928,264,9212,880,316,872,316,872,324,856,336,852,340,256,928,264,924,856,336,852,340,848,344,844,344,848,344,248,944,840,348,248,936,252,932,256,932,256,928,860,332,852,344,252,932,260,928,852,340,252,932,260,9216,876,320,868,324,860,332,856,332,856,340,256,928,260,924,856,340,852,340,848,344,844,344,844,352,248,932,848,344,252,936,256,928,260,932,260,920,860,332,856,340,252,932,260,928,852,340,256,932,256,9220,880,316,868,320,864,328,860,336,856,336,256,932,252,932,856,332,856,340,848,340,848,344,844,352,244,940,844,348,248,936,252,936,260,924,260,928,856,336,852,340,252,936,256,928,856,336,260,928,260,9220,876,320,864,324,868,324,860,336,856,332,260,932,256,928,852,340,852,340,848,344,848,344,840,348,252,936,848,348,248,940,252,928,260,928,264,924,856,336,848,344,252,932,260,928,856,336,256,932,256,9220,888,312,864,328,864,328,856,336,856,336,260,928,260,924,860,332,852,344,848,344,848,344,840,352,244,940,844,348,248,936,256,936,256,924,264,924,860,336,848,344,252,932,260,928,856,332,264,928,264,9216,884,312,868,320,868,328,860,336,852,336,256,932,260,928,856,336,856,336,852,340,848,344,848,344,248,940,848,344,248,940,252,936,256,928,260,924,856,340,852,340,252,940,256,924,856,336,260,928,264,9220,872,320,864,336,836,360,516,43624,28,10836,36,22024,144,36420,44,11332,36,7152,28,816,40,27832,48,18252,88,43744,24,42120,32,16368,36,240,56,104,32,184,48,632,28,43476,36,20132,60,12224,40,10636,40,14036,40,740,92,2408,32,12564,40,22364,36,6632,44,5552,32,5188,24,18152,40,192,32,15584,76,41324,28,8896,60,18724,24,18652,60,148,28,29268,24,14664,52,56752,48,240,36,30288,36,6728,88,16940,28,6684,52,21204,44,38280,24,13644,40,4248,20,564,32,636,48,15184,32,30648,32,9912,24,12236,56,16548,32,17224,20,96,76,376,24,396,60,60064,36,42112,44,14792,36,4560,44,236,24,15908,32,43496,28,2040,28,21316,40,200,100,4376,32,58796,20,
```
Transmitted signal from esp8266:
```
896,308,888,712,512,688,508,696,500,700,492,716,496,304,904,700,512,292,900,308,888,316,888,316,896,704,500,704,484,320,900,304,888,716,504,300,892,312,9312,704,468,732,492,704,484,720,488,716,516,288,924,280,880,724,488,712,524,680,496,708,524,676,516,288,936,668,512,292,892,316,904,300,880,324,888,716,492,712,504,300,876,328,880,724,492,308,888,316,9312,704,496,704,468,736,468,732,500,704,496,304,916,292,888,716,492,708,512,692,516,684,512,692,504,300,892,712,508,296,908,296,908,300,884,320,920,680,496,708,488,316,892,312,896,708,508,296,896,308,9272,744,468,732,472,728,488,712,484,720,492,312,880,320,876,732,484,716,500,704,500,704,516,684,496,308,912,692,504,304,896,308,900,300,896,312,912,692,524,680,496,308,888,316,912,692,504,296,896,308,9276,740,464,740,472,728,464,736,492,712,468,332,880,324,880,724,496,708,504,696,496,708,504,700,516,288,904,700,508,296,900,304,884,324,888,312,896,708,480,724,512,292,904,300,908,696,488,316,900,304,9284,728,472,732,476,724,484,716,480,724,492,312,912,292,888,716,488,712,536,668,504,700,496,704,516,288,908,696,504,300,900,308,888,316,888,316,892,712,496,704,508,300,904,300,892,708,500,304,896,308,9260,756,452,748,488,712,476,728,492,708,492,312,900,308,884,716,508,692,512,692,492,712,504,700,496,308,908,696,500,304,900,304,896,308,912,296,912,692,480,720,508,296,912,292,892,712,500,304,888,316,18924,32,7060,28,488,48,252,56,21940,40,1132,44,44380,28,23040,36,12104,24,43468,40,30324,44,27764,72,46684,32,48996,28,43652,44,20268,36,27080,32,37916,24,15840,60,488,24,1072,48,32300,44,54012,52,15476,48,132,32,35800,60,52,64,36116,28,992,28,676,28,36216,28,35888,104,124,96,28248,52,8888,52,19944,52,1512,40,124,52,13376,24,22732,36,15192,32,540,40,40992,32,60992,68,4220,56,16036,32,16604,48,3892,40,4660,20,12452,32,15304,60,21128,88,15848,68,100,36,23976,32,13492,48,68,20,52484,36,60936,20,43000,20,14492,40,52600,36,760,28,16220,52,36812,68,796,32,4236,44,16072,44,10184,28,10844,84,6680,44,44396,32,45748,44,22344,20,51492,80,40832,32,25464,28,16452,56,1332,40,9612,40,16136,36,
```
Still does not work. It needs a little more tweaking. I am trying to learn how to decode by myself but it's a bit complex for me.

Martin-Laclaustra, 2020-04-24:
394 may fit better than 400.
I do not think I can be of any more help. The signals are very similar already.
Did you try to transmit from the Arduino?
This repo's wiki ("Add New Remote" pages) has extensive and clear explanations on how all this works. Have you already studied them?

ninora, 2020-04-24:
I have a request to modify EV1527 timing and I am looking to hide someone for that.
Please email me at this gmail: ninorapub
Thanks

olonsoft, 2020-04-24:
Finally `{ 400, {  1, 23 }, {  1,  2 }, {  2,  1 }, false },` worked !!!
In fact, it works 8 in 10 times.
Comparing the two data I see that the width of the wider "ones" is a little less than the remote control. Can this be fixed?
Remote:
```
336,260,928,260,9216,884,308,876,320,864,328,860,328,860,336,260,928,260,924,860,332,856,336,852,340,848,344,844,348,248,936,848,344,252,932,256,932,260,928,260,924,860,332,856,336,260,928,260,924,860,332,264,924,264,9212,888,308,876,320,864,324,864,332,856,336,260,924,264,924,860,332,856,336,852,340,852,340,848,344,252,936,848,340,256,932,256,928,260,928,264,924,856,336,852,340,260,924,264,924,860,332,260,928,264,9212,888,312,872,320,868,324,864,328,860,332,260,928,264,924,860,332,856,332,852,340,852,340,848,348,252,932,848,344,256,932,256,928,260,928,264,924,860,332,856,336,256,928,264,924,860,332,264,924,264,9216,884,312,872,320,872,324,860,332,856,336,260,924,268,920,860,332,860,332,856,336,856,336,848,348,248,936,848,344,252,932,260,928,264,924,264,924,860,332,856,336,260,928,260,924,860,336,260,924,264,9220,884,308,876,320,868,324,864,328,860,332,264,924,264,920,864,332,848,344,824,372,668,8936,140,44128,44,20084,56,1528,84,29760,56,5456,28,51508,236,40,108,6444,80,7800,48,1432,56,16688,48,20688,108,3880,140,5164,40,11388,40,13000,52,4792,136,20200,56,12412,28,1880,36,28876,64,57028,124,224,36,40,40,13480,56,200,44,1472,84,29436,140,3472,60,38600,76,112,84,44,208,59572,68,84,88,5068,256,1820,48,14896,28,6104,104,30620,44,6264,36,13588,132,16784,64,6660,100,13300,28,1596,76,7868,72,3284,24,10424,40,13156,60,1944,28,8780,44,40800,72,52972,128,112,24,1528,92,19700,56,64,36,15012,60,3780,24,10984,48,6164,44,40,60,392,160,16532,32,27872,32,51048,24,6552,44,76,68,44,204,14760,56,31864,136,54860,200,24,28,616,32,14392,192,14080,60,96,292,520,32,19948,36,31364,200,57184,36,340,168,16884,88,19164,72,6340,92,8056,124,10636,32,5360,36,19800,72,164,36,1500,48,52,56,17960,140,9168,56,2428,40,13700,80,43460,104,76,36,13468,88,22764,92,304,52,31148,32,4836,36,2880,28,6432,60,42432,52,1808,36,28864,60,57572,48,45000,84,57264,88,13184,52,1720,68,1684,176,2072,24,108,64,22616,116,88,32,4220,48,49288,64,51440,32,788,36,56344,76,104,28,480,104,15472,92,47224,28,22272,128,10096,52,6276,52,328,48,13124,84,68,76,40,44,31232,76,5980,68,
```
ESP8266:
```
948,656,564,240,956,248,9320,692,528,672,520,680,544,660,540,664,560,244,944,260,948,652,560,644,552,652,556,644,564,640,560,244,960,644,560,244,964,240,956,252,948,256,952,648,560,644,584,220,952,252,960,644,544,264,940,264,9308,704,524,676,540,660,536,664,552,652,548,260,948,252,956,648,552,652,548,652,556,648,568,636,580,224,960,644,560,244,960,244,956,248,948,260,936,664,560,644,552,252,948,256,948,656,556,248,948,256,9320,692,528,680,524,672,540,660,552,652,568,236,960,244,960,640,548,660,544,656,560,640,560,644,560,244,964,640,556,248,956,252,956,248,948,256,948,656,548,656,560,240,952,256,948,656,556,244,948,260,9328,684,544,656,528,672,552,652,544,660,548,252,964,240,948,660,544,656,548,656,548,652,580,624,560,244,960,644,564,240,956,248,952,252,956,252,952,648,556,648,556,248,956,248,948,656,568,236,956,248,9320,692,524,676,544,660,548,656,544,656,540,264,952,252,940,664,556,644,564,640,560,644,552,648,564,244,956,644,572,232,964,244,944,260,956,248,940,664,548,652,564,244,964,240,956,648,552,252,952,252,9324,688,540,660,528,672,532,672,552,648,560,244,948,256,956,648,552,652,560,644,564,636,564,640,564,240,960,644,568,236,948,260,952,252,952,252,940,660,564,640,560,244,960,244,952,652,548,256,956,248,4660,48,43068,40,124,216,64,60,400,32,15848,68,20536,40,108,104,9424,40,49080,20,6144,36,13668,32,6104,152,1640,72,24892,84,27108,56,28688,44,228,128,17100,236,33216,48,51480,112,16556,40,1776,92,44996,56,39336,92,23144,36,18604,40,3820,32,20972,28,28704,40,6236,64,19688,60,35176,64,41064,40,4444,28,48308,60,49508,56,6704,32,50568,84,51148,100,46408,148,4424,184,51284,48,36,60,268,60,560,76,1000,36,8372,168,7844,36,28008,96,8676,24,47440,68,18136,112,4284,16,29892,48,49400,60,30496,40,6184,64,324,24,18112,112,11996,44,6692,32,46076,68,4424,32,4356,48,18988,48,144,96,512,32,15468,60,3924,20,29872,52,51820,20,18296,68,32112,100,76,28,51292,236,16628,84,9240,32,20244,140,42532,40,13528,104,18248,36,4644,108,13788,28,144,72,16364,44,1472,116,27916,64,50724,104,1560,76,3548,36,18816,36,27392,84,4484,64,51388,48,18872,28,27432,88,
```
But even if I let it as is, it's ok for me.
It would be perfect to know how did you find what settings it needed!
Thank you very very much !!!!!

Martin-Laclaustra, 2020-04-24:
Increase the number of repeats. Add an antenna of the correct length.
Again, the wiki has extensive explanations. Very short: bit duration (1 high + 1 low) is constant in your protocol. You can play around with the proportions between high and lows (e.g. 1:3 vs 2:3, current, 1:4 vs 3:4, initial) and describe the protocol pulse length in terms of the short part. The rest is using math.

olonsoft, 2020-04-24:
My STX882 is 50cm close to the receiver as I am testing now. I read the wiki again and again but I can not find how to group the numbers. In the example provided it's clear that are 3 groups. Can you please group for me the following?
```
24,24,24,24,24,24,28,28,28,28,28,28,28,28,28,32,32,32,32,32,32,32,32,36,36,36,36,36,36,36,36,36,36,36,36,40,40,40,40,40,40,40,40,40,44,44,44,44,44,44,44,44,44,48,48,48,48,48,48,48,48,52,52,52,52,52,52,52,56,56,56,56,56,56,56,56,56,60,60,60,60,60,60,60,64,64,64,64,64,68,68,68,68,68,72,72,72,72,76,76,76,76,76,76,76,80,80,84,84,84,84,84,84,88,88,88,88,88,92,92,92,92,96,100,104,104,104,104,108,108,108,112,112,116,124,124,128,128,132,136,136,140,140,140,140,160,164,168,176,192,200,200,200,204,208,224,236,248,248,252,252,252,252,256,256,256,256,256,256,256,260,260,260,260,260,260,260,260,260,260,260,260,260,260,260,260,260,260,260,264,264,264,264,264,264,264,264,264,264,264,264,264,264,264,264,268,292,304,308,308,308,312,312,320,320,320,320,320,324,324,324,324,328,328,328,328,328,332,332,332,332,332,332,332,332,332,332,332,332,332,332,332,332,332,336,336,336,336,336,336,336,336,336,336,336,336,336,340,340,340,340,340,340,340,340,344,344,344,344,344,344,348,348,348,372,392,480,520,616,668,788,824,844,848,848,848,848,848,848,848,848,848,852,852,852,852,852,852,856,856,856,856,856,856,856,856,856,856,856,860,860,860,860,860,860,860,860,860,860,860,860,860,860,860,860,860,864,864,864,864,864,864,868,868,872,872,872,876,876,876,884,884,884,888,888,920,920,924,924,924,924,924,924,924,924,924,924,924,924,924,924,924,924,924,924,924,928,928,928,928,928,928,928,928,928,928,928,928,928,932,932,932,932,932,932,936,936,936,1432,1472,1500,1528,1528,1596,1684,1720,1808,1820,1880,1944,2072,2428,2880,3284,3472,3780,3880,4220,4792,4836,5068,5164,5360,5456,5980,6104,6164,6264,6276,6340,6432,6444,6552,6660,7800,7868,8056,8780,8936,9168,9212,9212,9216,9216,9220,10096,10424,10636,10984,11388,12412,13000,13124,13156,13184,13300,13468,13480,13588,13700,14080,14392,14760,14896,15012,15472,16532,16688,16784,16884,17960,19164,19700,19800,19948,20084,20200,20688,22272,22616,22764,27872,28864,28876,29436,29760,30620,31148,31232,31364,31864,38600,40800,42432,43460,44128,45000,47224,49288,51048,51440,51508,52972,54860,56344,57028,57184,57264,57572,59572,
```

Martin-Laclaustra, 2020-04-24:
You probably posted a wrong set of timings. That is only a sequence of pulses with progressively increasing width.

olonsoft, 2020-04-25:
First of all, I want to thank you for the time you spend helping me.  
What I posted above is the sorted numbers of my original remote as the wiki says.
```
336,260,928,260,9216,884,308,876,320,864,328,860,328,860,336,260,928,260,924,860,332,856,336,852,340,848,344,844,348,248,936,848,344,252,932,256,932,260,928,260,924,860,332,856,336,260,928,260,924,860,332,264,924,264,9212,888,308,876,320,864,324,864,332,856,336,260,924,264,924,860,332,856,336,852,340,852,340,848,344,252,936,848,340,256,932,256,928,260,928,264,924,856,336,852,340,260,924,264,924,860,332,260,928,264,9212,888,312,872,320,868,324,864,328,860,332,260,928,264,924,860,332,856,332,852,340,852,340,848,348,252,932,848,344,256,932,256,928,260,928,264,924,860,332,856,336,256,928,264,924,860,332,264,924,264,9216,884,312,872,320,872,324,860,332,856,336,260,924,268,920,860,332,860,332,856,336,856,336,848,348,248,936,848,344,252,932,260,928,264,924,264,924,860,332,856,336,260,928,260,924,860,336,260,924,264,9220,884,308,876,320,868,324,864,328,860,332,264,924,264,920,864,332,848,344,824,372,668,8936,140,44128,44,20084,56,1528,84,29760,56,5456,28,51508,236,40,108,6444,80,7800,48,1432,56,16688,48,20688,108,3880,140,5164,40,11388,40,13000,52,4792,136,20200,56,12412,28,1880,36,28876,64,57028,124,224,36,40,40,13480,56,200,44,1472,84,29436,140,3472,60,38600,76,112,84,44,208,59572,68,84,88,5068,256,1820,48,14896,28,6104,104,30620,44,6264,36,13588,132,16784,64,6660,100,13300,28,1596,76,7868,72,3284,24,10424,40,13156,60,1944,28,8780,44,40800,72,52972,128,112,24,1528,92,19700,56,64,36,15012,60,3780,24,10984,48,6164,44,40,60,392,160,16532,32,27872,32,51048,24,6552,44,76,68,44,204,14760,56,31864,136,54860,200,24,28,616,32,14392,192,14080,60,96,292,520,32,19948,36,31364,200,57184,36,340,168,16884,88,19164,72,6340,92,8056,124,10636,32,5360,36,19800,72,164,36,1500,48,52,56,17960,140,9168,56,2428,40,13700,80,43460,104,76,36,13468,88,22764,92,304,52,31148,32,4836,36,2880,28,6432,60,42432,52,1808,36,28864,60,57572,48,45000,84,57264,88,13184,52,1720,68,1684,176,2072,24,108,64,22616,116,88,32,4220,48,49288,64,51440,32,788,36,56344,76,104,28,480,104,15472,92,47224,28,22272,128,10096,52,6276,52,328,48,13124,84,68,76,40,44,31232,76,5980,68,
```

> In practice: in excel: a) sort the durations; b) identify the groups; c) use the function "average" to average the durations within each group; d) approximate the greatest common divisor; e) check that the pulse length multiplied by the simple numbers roughly reproduce the initial timings.
What happens if there are more than 3 groups as that example has?

It also writes:
>"1" represented by a Low to High of 605 µs to Low = 1 * 605µs transmission "0" represented by a Low to High of 605 µs to Low = 2* 605µs = 1210µs transmission

How do you identify that "1" is [1*605LowToHigh+1*605ToLow] and "0" is [1*605LowToHigh+2*605ToLow]  and it's not the opposite?
Is it always "1" the first bit?  
Also, you suggest  `394, {  1, 23 }, {  1,  2 }, {  2,  1 }, false },` and then you write:
> You can play around with the proportions between high and lows (e.g. 1:3 vs 2:3, current, 1:4 vs 3:4, initial)

Where are 2:3 current and 3:4 initial?  
I also used [ProtocolAnalyzerDemo](https://github.com/Martin-Laclaustra/rc-switch/tree/protocollessreceiver/examples/ProtocolAnalyzeDemo) and the result was this `{ 297, {  1, 31 }, {  1,  3 }, {  3,  1 }, false }`
I am sorry but as you understand I am very confused.
Is there any other site where I can find examples?

olonsoft, 2020-04-25:
After a lot of hours of testing I made this:
```
{ 50,  { 10,182 }, { 10, 14 }, {  21, 3 }, false }
360,840,344,856,364,288,908,300,904,848,372,832,392,808,376,828,388,812,380,276,936,816,392,264,936,268,920,284,924,284,924,828,384,816,368,288,916,288,916,836,380,276,928,276,9280,892,320,872,352,852,348,848,372,832,372,280,924,280,912,844,368,832,380,824,408,796,384,816,416,240,944,808,392,264,924,280,932,276,932,272,912,840,392,808,376,280,932,272,920,836,372,280,936,268,9296,868,336,864,372,832,352,848,428,772,364,292,924,280,924,828,364,840,404,796,376,828,420,784,388,264,936,820,392,260,928,280,920,284,940,264,920,836,388,812,372,280,920,288,916,836,388,268,920,284,9280,884,352,848,352,848,360,844,380,820,364,288,924,284,924,828,380,820,388,816,388,812,420,784,388,268,932,824,392,260,936,268,928,280,912,292,924,828,372,828,388,268,920,288,932,816,400,256,940,264,9288,876,368,832,356,848,348,852,388,812,376,280,916,288,932,820,376,828,404,796,384,820,396,808,396,256,932,824,412,240,936,272,932,272,920,284,932,820,380,824,368,284,928,280,928,824,372,284,936,268,9284,880,332,868,412,788,344,860,384,816,356,296,940,268,908,844,392,808,380,824,420,784,376,824,412,244,924,828,420,236,940,264,916,288,932,276,912,840,412,788,364,292,928,276,932,820,368,288,916,288,9288,880,360,836,340,860,372,832,360,840,404,252,920,284,928,824,396,808,364,836,412,792,380,824,400,256,932,820,392,260,944,264,940,264,928,276,932,820,408,796,380,272,916,288,956,800,388,264,924,284,20104,1116,80,284,56,208,3976,32,31148,516,36,396,24,448,96,220,35084,676,76,224,68,248,40,136,48,68,56664,44,30456,80,13564,996,44,360,128,100,46792,240,53760,1336,35472,1200,80,108,64,48,196,100,140,84,92,84,216,88,260,60,4160,44,16328,52,13496,732,60,532,76,68,36,136,84,20,35112,632,48,740,164,60,35220,452,36,660,152,168,92,144,35192,940,32,276,76,84,144,120,324,524,180,52,52,32,184,52,112,96,708,108,1564,96,220,84,160,80,3688,28,42624,24,60,48,48392,28,1144,1000,40,124,48,400,136,52,35100,440,28,472,76,528,84,132,136,120,56,52,172,72,28,48,88,292,164,112,156,44,296,96,224,124,420,68,48336,28,
```
Which is almost identical to what my remote control transmits:
```
336,260,928,260,9216,884,308,876,320,864,328,860,328,860,336,260,928,260,924,860,332,856,336,852,340,848,344,844,348,248,936,848,344,252,932,256,932,260,928,260,924,860,332,856,336,260,928,260,924,860,332,264,924,264,9212,888,308,876,320,864,324,864,332,856,336,260,924,264,924,860,332,856,336,852,340,852,340,848,344,252,936,848,340,256,932,256,928,260,928,264,924,856,336,852,340,260,924,264,924,860,332,260,928,264,9212,888,312,872,320,868,324,864,328,860,332,260,928,264,924,860,332,856,332,852,340,852,340,848,348,252,932,848,344,256,932,256,928,260,928,264,924,860,332,856,336,256,928,264,924,860,332,264,924,264,9216,884,312,872,320,872,324,860,332,856,336,260,924,268,920,860,332,860,332,856,336,856,336,848,348,248,936,848,344,252,932,260,928,264,924,264,924,860,332,856,336,260,928,260,924,860,336,260,924,264,9220,884,308,876,320,868,324,864,328,860,332,264,924,264,920,864,332,848,344,824,372,668,8936,140,44128,44,20084,56,1528,84,29760,56,5456,28,51508,236,40,108,6444,80,7800,48,1432,56,16688,48,20688,108,3880,140,5164,40,11388,40,13000,52,4792,136,20200,56,12412,28,1880,36,28876,64,57028,124,224,36,40,40,13480,56,200,44,1472,84,29436,140,3472,60,38600,76,112,84,44,208,59572,68,84,88,5068,256,1820,48,14896,28,6104,104,30620,44,6264,36,13588,132,16784,64,6660,100,13300,28,1596,76,7868,72,3284,24,10424,40,13156,60,1944,28,8780,44,40800,72,52972,128,112,24,1528,92,19700,56,64,36,15012,60,3780,24,10984,48,6164,44,40,60,392,160,16532,32,27872,32,51048,24,6552,44,76,68,44,204,14760,56,31864,136,54860,200,24,28,616,32,14392,192,14080,60,96,292,520,32,19948,36,31364,200,57184,36,340,168,16884,88,19164,72,6340,92,8056,124,10636,32,5360,36,19800,72,164,36,1500,48,52,56,17960,140,9168,56,2428,40,13700,80,43460,104,76,36,13468,88,22764,92,304,52,31148,32,4836,36,2880,28,6432,60,42432,52,1808,36,28864,60,57572,48,45000,84,57264,88,13184,52,1720,68,1684,176,2072,24,108,64,22616,116,88,32,4220,48,49288,64,51440,32,788,36,56344,76,104,28,480,104,15472,92,47224,28,22272,128,10096,52,6276,52,328,48,13124,84,68,76,40,44,31232,76,5980,68,
```
Now my device resonds 100% to my commands.
Thank you again @Martin-Laclaustra  for your time.

olonsoft, 2020-04-29:
FYI: My new settings work, but I wanted to find what was wrong.
There must be a hardware problem with this transmitter module.  
Today I went to my lab and analyzed the signals at the transmitter pin and receiver pin.
I noticed that the width of the transmitted pulses is shorter than what I really send. 
Are average 125μs shorter. 
[image]
I 'll try to modify the source to transmit +125μs to highs and -125μs to lows and see how it goes.