sui77/rc-switch #336: Sends only protocol 1, nothing sent for protocol 2 (or other)

sunbee, 2020-05-03:
I have set up a transmitter and receiver on one Arduino Uno. The send and receive work correctly for protocol 1, bit-length 24, with examples similar to those in the repo. When I change transmission settings to use protocol 2 or bit-length 32 with the set methods, nothing appears to be transmitted. I have a separate remote-control transmitter from BN-LINK that transmits codes over 433.92 MHz using protocol 2 and bit-length 32. The remote works correctly, as evidenced by switching the hardware (relay) on/off as well as reception of the numeric codes and associated parameters (protocol, bit-length) at the Arduino. So this library lets me hook in the BN-LINK codes but does not appear to support transmission to clone the remote with Arduino.

Any help is appreciated, thanks.

Empire99, 2020-05-04:
I have the same issue.

1technophile, 2020-05-04:
You are using the current master or a special version?

Empire99, 2020-05-04:
I am using rc-switch by sui77 version 2.6.3, that is installed by arduino library manager.

The code I use for sending is:

```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
    Serial.begin(115200);
   
    mySwitch.enableTransmit(5);
    mySwitch.setProtocol(2);  
}

void loop() {
    mySwitch.send(3074463891, 32);
    delay(1000);
    mySwitch.send(270166227, 32);
    delay(3000);
}
```

1technophile, 2020-05-04:
Did you uploaded on another board with a receiver and https://github.com/sui77/rc-switch/tree/master/examples/ReceiveDemo_Advanced to check what has been sent?

Empire99, 2020-05-04:
Thanks for your advice, I will debug it using another board.