sui77/rc-switch #339: Support for chipset FHT7901T Ver1.1

crainsaw, 2020-05-07:
Hello,
I have bought [Meister Funksteckdosen-Set 3+1, 3680 W, IP20, 7474230](https://www.amazon.de/Meister-Funksteckdosen-Set-3680-IP20-7474230/dp/B016UVYMWO) and they use the chipset [FHT7901T](http://www.datasheet39.com/function/parts.php?query=FHT7901T) Ver1.1 for 433 MHz communication.

Is it possible to add support for this chipset as well?
If not I would be happy about a tutorial link for how to reverse engineer such a protocol.

Thanks in advance!

holesond, 2020-06-27:
After looking at the FHT7901T Ver1.1 PCB, I think the board might be using an LX2262 encoder (http://en.chiptrue.com/images/LX2262_en.pdf), which is said to be compatible with PT2262. Furthermore, FHT-7901 is mentioned several times in https://github.com/sui77/rc-switch/wiki/List_KnownDevices. Thus I guess it could already be supported by rc-switch. If that is not the case, Li-Wen Yip wrote a tutorial on decoding the signals of LX2262 and some other encoder, see https://www.liwen.id.au/arduino-rf-codes/.

crainsaw, 2020-07-19:
> After looking at the FHT7901T Ver1.1 PCB, I think the board might be using an LX2262 encoder (http://en.chiptrue.com/images/LX2262_en.pdf), which is said to be compatible with PT2262. Furthermore, [...]

Thank you very much for the info! Its not currently working with RC-Switch. I will try the mentioned solution from Li-Wen Yip.

1technophile, 2020-07-19:
Did you tried to follow this tutorial?
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1

holesond, 2020-07-19:
Recently, I tried controlling an FHT7901T outlet with RC-Switch. 

Firstly, I tried protocol 1, pulse width 150 (suggested in List_KnownDevices) and the switchOn/switchOff functions for "Type A: 10 pole DIP switches" and "Type B: Two rotary/sliding switches" mentioned in https://github.com/sui77/rc-switch/wiki/HowTo_OperateLowCostOutlets. These did not work with my outlet DIP switch code 10000 10000 (1-on, 0-off). 

Secondly, I used a 433 MHz receiver and the ReceiveDemo_Simple example from rc-switch to receive the bit sequence transmitted by the original remote control transmitter. For the DIP switch code 10000 10000, the transmitted sequences were (decimal data encoding): 

- ON signal: Received 13976916 / 24bit Protocol: 1
- OFF signal: Received 13976913 / 24bit Protocol: 1

These codes transmitted via RC-Switch work, the outlet can be switched on and off.
Setup commands used:
`mySwitch.setProtocol(1); `
`mySwitch.setPulseLength(150);`
Control commands:
`mySwitch.send(13976916, 24); // switch on `
 `mySwitch.send(13976913, 24); // switch off `
(I have not tried adding this FHT7901T as a new device to the rc-switch library yet.)