sui77/rc-switch #347: Detect Signal Code and Send Again Not Work- Protocol Analyzer

spamws, 2020-05-21:
Hi
I am using protocol analyzer and detect 315mhz remote protocol with code then resend decoded signal again, but the device not work.
I check my send signal with protocol analyzer and result is the same but remote control work properly and my send code not work.
My Simple Rc Scanner Code is : 
First:
`118,84,153,90,72,162,178,142,362,195,824,243,43,116,78,173,372,368,756,251,70,77,138,85,55,120,181,631,446,61,70,270,795,150,442,78,414,147,1013,112,1091,336,162,69,97,181,564,359,171,401,157,229,818,229,641,192,36,145,790,215,56,78,165,283,1386,80,204,294,65,76,227,575,333,116,501,183,111,63,34,131,304,74,141,294,66,144,34,408,237,236,675,344,42,124,533,105,59,149,412,204,290,148,1181,308,96,73,229,272,176,146,373,72,437,218,277,277,117,221,96,180,191,125,62,158,247,278,32,329,408,207,66,161,360,208,52,181,208,144,74,95,304,395,803,279,69,154,161,67,274,351,568,1131,1149,1168,2635,64,1062,47,2817,94,1495,65,360,68,1633,107,51,90,63,35,970,70,42,58,843,122,866,101,876,169,1056,284,548,361,1416,277,446,219,238,115,716,349,757,68,231,175,1100,146,1334,100,106,46,597,101,75,88,146,284,1157,176,494,340,264,139,151,55,668,163,185,115,52,203,72,292,508,91,77,68,319,117,765,191,701,48,38,317,1174,155,628,107,236,122,94,247,608,244,341,224,197,157,427,251,197,108,476,110,213,95,45,113,113,159,69,76,194,94,407,335,600,199,76,106,43,269,55,72,241,144,142,188,209,80,633,157,224,27,493,457,263,182,294,465,55,142,465,902,66,75,19,526,533,904,169,394,656,894,173,350,715,380,671,880,189,856,203,361,701,872,197,852,208,855,206,320,743,306,748,324,747,297,765,293,766,301,602,987,234,822,244,294,761,303,756,297,8186,297,766,828,235,290,769,828,236,289,771,826,241,288,766,292,768,824,238,820,242,283,785,806,256,798,265,806,255,268,799,265,785,277,336,977,537,272,410,1046,396,795,261,801,263,267,793,259,797,271,8205,299,763,825,237,298,763,824,252,258,800,783,277,264,796,261,800,789,273,791,269,257,806,796,261,807,253,796,268,258,798,282,783,282,775,268,793,279,781,288,778,807,250,811,257,262,799,268,793,282,8188,295,770,825,237,295,766,824,234,289,770,824,239,287,776,286,771,827,240,808,249,285,782,821,240,819,235,829,236,283,780,272,788,280,777,286,778,284,776,280,781,827,238,804,259,280,778,292,763,300,8184,303,760,831,230,`
Second:
`293,769,282,249,819,241,834,235,819,238,816,250,802,262,786,799,277,787,288,240,824,237,803,258,8202,333,727,857,214,314,752,1368,750,836,228,306,753,305,756,835,230,829,252,239,817,779,284,769,294,778,282,239,819,251,805,256,815,259,802,254,805,260,801,776,275,797,276,255,803,258,797,259,3257,983,3962,327,744,842,215,318,749,840,223,302,755,840,226,294,765,304,758,835,223,834,232,288,777,828,227,798,275,779,288,248,810,252,807,257,816,246,806,255,813,246,813,788,271,785,267,255,805,259,807,263,8217,278,777,827,244,273,782,822,241,285,777,819,245,262,800,271,784,827,236,808,250,283,780,805,259,800,285,788,266,243,805,265,807,257,797,265,769,321,750,311,748,842,225,833,225,302,767,297,757,301,8182,316,744,847,220,307,751,1371,756,824,244,264,798,252,804,801,274,790,262,269,799,783,277,791,261,799,271,256,805,255,837,235,786,266,800,261,811,252,793,803,265,808,252,270,794,263,803,278,3357,1101,3732,311,750,842,218,310,750,841,221,307,761,831,226,302,760,302,759,836,227,813,288,223,809,782,287,778,274,784,277,263,812,234,813,264,809,247,808,257,797,266,791,803,260,800,270,257,803,266,785,269,8199,333,729,854,219,310,747,835,234,306,753,835,232,306,752,304,751,838,234,818,253,259,807,786,277,785,273,777,280,259,800,261,806,251,799,266,801,269,790,261,799,804,278,776,268,264,788,270,804,266,8190,330,732,853,218,314,745,845,226,303,752,1367,762,297,761,831,229,833,251,242,820,780,286,770,288,769,284,258,800,258,808,254,808,259,814,236,820,251,795,798,266,796,265,259,802,256,815,253,4605,53,20,1000,2536,300,769,815,243,293,767,796,270,268,783,832,237,277,788,268,788,817,247,804,266,262,798,802,252,813,249,801,265,267,796,265,793,264,792,279,796,273,784,281,778,812,254,804,248,276,802,250,789,279,8189,329,733,858,215,314,750,843,219,307,747,840,228,302,759,302,757,842,219,837,228,295,769,827,262,767,284,769,293,256,805,249,826,233,822,238,810,257,792,277,804,790,273,784,274,252,805,259,795,269,8194,327,733,859,218,312,743,843,221,311,195,1391,230,293,773,288,768,839,223,824,241,291,764,`
And Third:
`807,264,810,245,8200,334,729,860,218,306,753,831,228,308,758,834,226,290,770,304,750,840,228,830,235,290,770,830,233,818,266,757,300,245,817,241,815,249,820,231,817,247,816,247,811,781,277,793,282,246,815,249,803,256,7971,1018,280,846,223,302,759,830,231,306,755,832,232,282,777,296,766,829,230,834,229,285,772,831,232,823,268,760,220,962,177,228,826,233,837,229,825,234,822,239,814,782,291,780,282,241,825,236,837,235,8197,326,746,846,216,309,750,840,225,308,759,828,230,293,765,306,756,832,228,833,232,296,766,822,257,777,283,776,294,237,834,226,825,233,830,230,814,250,807,256,803,794,284,761,289,257,822,235,807,266,8193,328,735,856,211,318,745,843,222,307,756,835,224,297,769,287,770,834,230,831,230,288,773,824,236,828,262,757,318,214,823,240,817,244,824,247,810,253,803,255,800,790,279,775,291,249,812,252,799,265,8194,1029,46,843,216,305,759,836,228,297,765,830,229,292,770,305,760,829,235,826,232,287,771,820,297,733,292,774,294,1298,819,241,819,242,817,243,818,245,808,783,295,776,287,239,826,239,814,238,8207,332,732,855,216,303,764,833,219,314,755,833,234,300,758,300,757,832,231,835,232,280,779,825,261,769,279,770,299,246,808,247,812,248,816,254,801,260,806,261,790,790,285,777,273,256,812,246,814,249,8204,332,731,860,209,316,750,844,221,305,756,837,223,297,763,306,756,838,225,836,226,292,774,818,254,784,275,773,284,259,807,255,812,243,817,248,808,260,806,247,807,787,281,781,278,261,801,254,813,253,8195,1914,223,305,758,837,217,310,754,833,229,294,772,290,770,831,227,828,235,285,777,823,258,762,297,784,307,254,38,1011,790,247,825,236,817,238,824,241,820,780,272,776,283,256,814,240,822,243,8208,328,732,851,224,311,748,842,216,311,758,832,230,297,769,288,763,838,223,838,234,287,769,827,262,763,297,769,274,259,808,244,820,235,824,251,812,245,836,227,805,785,285,786,274,253,809,253,797,272,8193,327,737,852,221,310,747,843,225,304,762,826,230,294,766,301,759,839,220,837,231,289,773,818,280,749,285,784,286,241,817,252,817,235,867,201,830,234,811,248,808,790,266,792,267,263,800,256,812,257,`

My Sending Protocol:
`Received 5561356 / 24bit Protocol: 1
data bits of pulse train duration: 25461
proposed protocol: { 265, { 1, 31 }, { 1, 3 }, { 3, 1 }, false }`

I am sending 5561356 with  { 265, { 1, 31 }, { 1, 3 }, { 3, 1 }, false } but device not work.
Please help me to find what is my fault.
Thanks in advance. 
Note : This remote and device not work with this process. Other worked.
Thanks @Martin-Laclaustra

Martin-Laclaustra, 2020-05-21:
You can go through the steps described in:
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_1
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2

You will see there is a step in which you compare your original and your microcontroller-sent signals, recorded with Simple Rc Scanner.
First suggested check is: are you sending 24 bits?

spamws, 2020-05-21:
Thanks for your replay. 
I read those links previously but when i use your protocol analyzer get same result with original remote send and my send. 
Yes i send 24 bits.

Martin-Laclaustra, 2020-05-29:
Similar output in protocol analyzer is no guarantee.
https://github.com/sui77/rc-switch/wiki/Add_New_Remote_Part_2#if-your-changes-dont-work