sui77/rc-switch #351: Support to HCS301 chips

EFXPROMedia, 2020-06-15:
I was wondering if there is a possible way to replicate the functionality of the HCS301 chips, I have a gate that I would like to automate with an esp8266 but I have tried all possible libraries to get the code but apparently based on an article I found [here](http://lmpautomation.com/blog/?p=24), HCS301 "Automatic Encoding, using algorithm to generate different code every time".

I also imagine that I cannot use a clone RF receiver you get on Aliexpress and clone the code to another chip and extract it from there using RC-Switch would it be possible?

VICTOROUOU, 2022-07-23:
Удалось ли вам в итоге воспроизвести код эмитента HCS301?

VICTOROUOU, 2022-07-23:
Were you able to reproduce the HCS301 issuer code in the end?