sui77/rc-switch #359: Error while calling RCSwitch::switchOn

DeveloperTK, 2020-08-13:
Hi, I am currently writing a program, where an RCSwitch can be controlled by HTTP Requests. I wrote some code in order to parse the GET Argument:

    String header; // HTTP-Header
    
    char ht[header.length()];
    header.toCharArray(ht, header.length());
    char* firstLine = strtok(ht, "\n");
    char* line2 = firstLine + 4;
    char* query = strtok(line2, " ");
    char* params = strtok(query, "/");
    String par2 = String(strtok(params, "?"));

    // par2 looks something like like this: 11111000101
    
    String house = par2.substring(0, 5); // in this case: 11111
    String device = par2.substring(5, 10); // in this case: 00010
    String state = par2.substring(10); // in this case 1

Then I convert the strings to `char*` and try to call the `RCSwitch::switchOn` function:

```
char* house2;
char* device2;
char* state2;

house.toCharArray(house2, 5);
device.toCharArray(device2, 5);
state.toCharArray(state2, 5);

if(state == "1") {
  trans.switchOn(house2, device2);
} else {
  trans.switchOff(house2, device2);
}
```

Then somehow, it spits out this:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2bf2  PS      : 0x00060430  A0      : 0x800d2d6d  A1      : 0x3ffb1cd0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x3ffb1f68  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000005  
A10     : 0x3ffc11b8  A11     : 0x3f401663  A12     : 0x00000030  A13     : 0x00000046  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d2bf0  LEND    : 0x400d2c06  LCOUNT  : 0x00000004  

Backtrace: 0x400d2bf2:0x3ffb1cd0 0x400d2d6a:0x3ffb1cf0 0x400d1406:0x3ffb1d10 0x400d46e5:0x3ffb1fb0 0x40089499:0x3ffb1fd0
```

which translates to this:

```
PC: 0x400d2bf2: RCSwitch::getCodeWordA(char const*, char const*, bool) at /Users/.../Documents/Arduino/libraries/rc-switch-2.6.3/RCSwitch.cpp line 300
EXCVADDR: 0x00000000

Decoding stack results
0x400d2bf2: RCSwitch::getCodeWordA(char const*, char const*, bool) at /Users/.../Documents/Arduino/libraries/rc-switch-2.6.3/RCSwitch.cpp line 300
0x400d2d6a: RCSwitch::switchOn(char const*, char const*) at /Users/.../Documents/Arduino/libraries/rc-switch-2.6.3/RCSwitch.cpp line 277
0x400d1406: loop() at /Users/.../Documents/Arduino/espRFWebControl/espRFWebControl.ino line 141
0x400d46e5: app_main() at /Users/.../Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 26
0x40089499: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

(espRFWebControl.ino line 141 is `trans.switchOn(house2, device2);`)

1technophile, 2020-08-13:
Hi,

Did you tried to serial print the result of:
```
house.toCharArray(house2, 5);
device.toCharArray(device2, 5);
state.toCharArray(state2, 5);
```

before injecting it into RCSwitch commands?

DeveloperTK, 2020-08-13:
yes that gave me 
```
1111
00010
1
```
as expected

1technophile, 2020-08-13:
Declare only the char array size that you need:
```
char* house2;
char* device2;
char* state2;
```
And be careful about [null terminator](https://classes.mst.edu/compsci1570/ntca.htm)

DeveloperTK, 2020-08-13:
Thanks, initializing the char arrays with a set length first fixed the exception. But now nothing happens. The printed values are still as expected and manually executing the command turns the switch on/off. Only my dynamic solution does nothing...

1technophile, 2020-08-13:
>  manually executing the command turns the switch on/off.

Could you elaborate on what do you mean by manually, please?

Could you post the whole code also?

DeveloperTK, 2020-08-13:
> And be careful about null terminator

Thank you that resolved the problem. I ended up doing this:
```
char house2[6];
char device2[6];
char state2[2];

house.toCharArray(house2, 6);
device.toCharArray(device2, 6);
state.toCharArray(state2, 2);
```

closed 2020-08-13 by DeveloperTK