sui77/rc-switch #364: no input on receiver over 725 microseconds pulse lenght

zoneamick, 2020-10-10:
Hello,
i did found the rc-switch repository to use my rc remote which is quite nice but my first project wont work maybe im doing something wrong or i got bad hardware. I use a 315mhz transmitter/combo from ebay.

#### Receiver input from my remote "ON/OFF" received on an arduino
```
Decimal: 13328 (15Bit) Binary: 011010000010000 Tri-State: not applicable PulseLength: 842 microseconds Protocol: 5
Raw data: 11796,548,1828,1748,640,1748,644,548,1832,832,108,16,1828,548,1828,552,1828,548,1828,552,1832,1740,644,552,1824,552,1832,544,1828,552,1832,

Decimal: 13320 (15Bit) Binary: 011010000001000 Tri-State: not applicable PulseLength: 842 microseconds Protocol: 5
Raw data: 11792,552,1832,1744,640,1748,644,548,1828,1748,644,548,1832,544,1832,548,1828,548,1832,548,1832,544,1832,1744,644,548,1832,548,1832,544,1832,
```

#### I use this sketch to transmit using an arduino and an ESP8266
```
#include <RCSwitch.h>
RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  
  mySwitch.enableTransmit(5);
  pinMode(LED_BUILTIN, OUTPUT);

    mySwitch.setProtocol(5,844);
  // Optional set number of transmission repetitions.
  //mySwitch.setRepeatTransmit(15);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH); 
  mySwitch.send("011010000010000");
  delay(5000);  
  digitalWrite(LED_BUILTIN, LOW); 
  mySwitch.send("011010000001000");
  delay(5000);
}
```

The tests i've done:
1. Use the arduino IDE for the arduino and the ESP8266 same results both over 725 microseconds no output on the receiver
2. Use tasmota-sensors on the esp8266 same problem, i did use the console to RFSend
3. swapping transmitter model, i get a wrong protocol on the receiver even on low microseconds
4. Try an 9" antenna, with or without
5. uncommented setRepeatTransmit to 15

here's the devices i used:
[image]

Thanks!

1technophile, 2020-10-12:
The photos confuse me, if you have the setup working with one transmitter, this may due to a hardware problem on the 2 other ones ?

zoneamick, 2020-10-17:
> The photos confuse me, if you have the setup working with one transmitter, this may due to a hardware problem on the 2 other ones ?

what i want to say it's the first transmitter (on the picture) when i use pulselenght 720 it works over 725 i cant get any transmission.