sui77/rc-switch #365: how to simple on off stream binary

juankr10s, 2020-10-10:
hi community
i need sen a simple on off serial binary stream
without protocols 
only on off in a time secuence 
is posible in rc-switch?

juankr10s, 2020-10-11:
i try edit protocol : 

{ 300, {  1, 1 }, {  0,  1 }, {  1,  0 } },    // protocol 5
but bit "0" is 96us more long than "1"
[image]

zoneamick, 2020-10-11:
as a simple question do you use a 315mhz transmitter?

juankr10s, 2020-10-11:
hi
no, 434Mhz (rf module 433mhz)

1technophile, 2020-10-12:
> without protocols

It is not possible without a protocol, the protocol defines the signal modulation and demodulation.

But you can stick with protocol 1 and consider a decimal value 1234567 as ON and a decimal value 7654321 as OFF.

juankr10s, 2020-10-12:
mmm ok
i think in a loop with delay (ms) on off gpio port for rf 433 module
well
thank you

closed 2020-10-12 by juankr10s

1technophile, 2020-10-12:
There is also other protocols that support ON OFF like the one below:
https://github.com/sui77/rc-switch/blob/master/examples/TypeA_WithDIPSwitches/TypeA_WithDIPSwitches.ino if you don't want to reinvent the wheel.

> i think in a loop with delay (ms) on off gpio port for rf 433 module

Be careful about not jamming the 433mhz frequency