sui77/rc-switch #366: Receiving custom protocol PIR's and contacts for MQTT

SparkyDan555, 2020-10-12:
I am trying to receive the 433.92 MHz signals from my PIR's and door contacts and publish their states over MQTT on an ESP8266 with RXB6 superhet module. I have used RecieveDemo_Advanced.ino to decode the signals from the PIR's but the door contacts do not show up. For these I have had to use the modified library from [https://github.com/Martin-Laclaustra/rc-switch/blob/protocollessreceiver/examples/ProtocolAnalyzeDemo/ProtocolAnalyzeDemo.ino](url) which works successfully and gives me the custom protocol { 620, { 1, 8 }, { 1, 2 }, { 2, 1 }, true } for the door contacts and { 610, { 1, 10 }, { 1, 2 }, { 2, 1 }, true } for the PIRs. From here I don't know where to go to integrate the RC-Switch code into my MQTT sketch. I need to use custom protocols to make them work but I don't know if I can use custom protocols for receiving, or if this is just for if I were to emulate the code with a transmitter? I am not interested in transmitting, I just need to receive the codes using the previously mentioned protocols so I can update my MQTT topics. Any help appreciated. Thanks!

1technophile, 2020-10-12:
Take a look to [OpenMQTTgateway](https://github.com/1technophile/OpenMQTTGateway) it does already this job.
Just add the protocol to the rcswitch library once downloaded.

SparkyDan555, 2020-10-12:
Thanks for the reply. I'm a little confused by the documentation, I have changed void setup() of RecieveDemo_Advanced.ino to:

`void setup() {
  Serial.begin(9600);
  static const RCSwitch::Protocol customprotocol = { 620, { 1, 8 }, { 1, 2 }, { 2, 1 }, true };
  mySwitch.setProtocol(customprotocol);
  mySwitch.enableReceive(D2);  // Receiver on interrupt 0 => that is pin #2
}`

but when I trigger my door contacts nothing shows up in the serial monitor? I did have a look at OpenMQTTGateway but my specific application requires some processing of the RF codes and the MQTT topics need to be published in a very specific way, plus I'd rather learn something new and learn to do this myself.

1technophile, 2020-10-12:
Did you try to set the protocol hardcoded [here](https://github.com/sui77/rc-switch/blob/ebe91719936265540adcb9766f6ed6e09befe48f/RCSwitch.cpp#L80) ?

SparkyDan555, 2020-10-12:
Just tried this, no luck unfortunately.