sui77/rc-switch #369: ICACHE_RAM_ATTR missing for interrupt handler?

fcassirer, 2020-10-28:
I just installed this library and with a simple setup() which simply called:

RCSwitch mySwitch = RCSwitch();

void setup() {
 mySwitch.enableReceive(D3);
}

I got a panic:

17:25:11.720 -> Starting setup
17:25:11.720 -> **ISR not in IRAM!**
17:25:11.720 -> 
17:25:11.720 -> User exception (panic/abort/assert)
17:25:11.720 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:25:11.758 -> 
17:25:11.758 -> Abort called
17:25:11.758 -> 
17:25:11.758 -> >>>stack>>>
17:25:11.758 -> 
17:25:11.758 -> ctx: cont
17:25:11.758 -> sp: 3ffffed0 end: 3fffffc0 offset: 0000
17:25:11.758 -> 3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
17:25:11.758 -> 3ffffee0:  000000fe 00000000 00000000 00000000  
17:25:11.758 -> 3ffffef0:  00000000 00000000 00000000 00ff0000  
17:25:11.758 -> 3fffff00:  5ffffe00 5ffffe00 00000020 00000000  
17:25:11.758 -> 3fffff10:  00000003 00000000 3ffee504 402022d6  
17:25:11.758 -> 3fffff20:  401003fa 3ffef36c 3ffef354 402022e8  
17:25:11.758 -> 3fffff30:  3ffe8620 00000000 3ffee504 402027fd  
17:25:11.786 -> 3fffff40:  00000000 0e99c9bb ffffff00 00000001  
17:25:11.786 -> 3fffff50:  4010182d 000e78be 3ffee5d8 3ffee57c  
17:25:11.786 -> 3fffff60:  3fffdad0 00000000 3ffee504 402028ac  
17:25:11.786 -> 3fffff70:  402014d8 3ffee504 3ffe8610 40201108  
17:25:11.786 -> 3fffff80:  3fffdad0 00000000 3ffee504 4020111c  
17:25:11.786 -> 3fffff90:  3fffdad0 00000000 3ffee504 4020105c  
17:25:11.819 -> 3fffffa0:  feefeffe feefeffe 3ffee53c 40201ee4  
17:25:11.819 -> 3fffffb0:  feefeffe feefeffe 3ffe84e4 40100cf9  
17:25:11.819 -> <<<stack<<<

This is for a nodemcu using arduino boardmgr 2.7.4 and RCSwitch v 2.6.3

Looking at the code, it seems the interrupt handler for the initial enable needs to have ICACHE_RAM_ATTR:

765:  `void ICACHE_RAM_ATTR RCSwitch::handleInterrupt() {`

added to insure the interrupt handler is in the proper memory space (I believe ...)

Added this locally and it worked.  

I am guessing this might need to be conditional (#ifdef'd)  to support all the various platforms.

1technophile, 2020-10-28:
This issue is corrected on the [master](https://github.com/sui77/rc-switch/commit/ebe91719936265540adcb9766f6ed6e09befe48f)

fcassirer, 2020-10-28:
Ah, my bad, sorry, looks like there was an older version of the library I had downloaded a while back.

closed 2020-10-28 by fcassirer