sui77/rc-switch #372: Dooya 2702 RF Remote

martin072, 2020-11-08:
Hi All, 

Did any one managed to get the 15 Channel Dooya 2702 RF remote working with this library (https://www.aliexpress.com/i/32917187487.html)? I have the device in house, but it does not seem to work. Some guidance how to sniff the remote would be an option as well so I can try to get the signals?

Many thanks

martin072, 2020-11-19:
So I've managed to run the scanner, however, I am a bit stuck with transposing all the values... if there is someone who could help me out.. I looked at the 'Add new remote' section, but a bit confused how to interpret the date.. I've included two samples here:

One:
1076,1426,13207,2188,1072,1062,1057,1073,719,1428,720,1090,1075,1428,1073,1072,718,1077,1074,1090,1074,1074,1076,1071,1428,1070,1071,735,1076,1425,1070,720,1075,1076,1426,732,1074,1076,1426,719,1073,1074,1429,13200,2191,1066,1064,1060,1073,719,1425,721,1088,1076,1427,1070,1073,719,1073,1076,1087,1075,1074,1076,1072,1424,1071,1071,736,1075,1422,1074,718,1073,1073,1426,734,1076,1072,1425,719,1077,1073,1426,13201,2192,1068,1065,1056,1075,717,1426,720,1088,1076,1424,1072,1072,718,1078,1072,1088,1077,1072,1077,1071,1426,1070,1072,734,1077,1422,1075,716,1074,1075,1427,732,1075,1076,1422,720,1073,1076,1427,13200,2189,1070,1059,1064,1074,720,1423,721,1089,1075,1429,1069,1071,720,1075,1076,1087,1077,1072,1076,1074,1425,1072,1069,736,1076,1424,1072,717,1076,1075,1424,734,1077,1073,1426,719,1073,1077,1428,13206,2191,1066,1056,1068,1071,722,1428,718,1089,1075,1428,1071,1071,724,1070,1076,1091,1073,1072,1075,1076,1425,1072,1070,738,1074,1423,1073,721,1073,1075,1424,735,1077,1073,1429,717,1073,1076,1424,13443,2187,1072,1053,1070,1070,721,1426,720,1089,1074,1426,1075,1067,721,1074,1077,1085,1074,1076,1073,1075,1423,1071,1075,731,1074,1425,1071,720,1075,1075,1424,732,1077,1073,1427,716,1075,1077,1422,13057,2188,1071,1071,1050,1076,717,1428,722,1090,1073,1427,1072,1073,719,1077,1074,1090,1075,1074,1074,1075,1426,1071,1071,734,1076,1424,1072,718,1077,1075,1427,732,1076,1073,1427,719,1076,1074,1429,13152,2191,1067,1072,1055,1069,721,1429,721,1087,1075,1425,1073,1074,719,1073,1077,1091,1073,1074,1075,1074,1428,1069,1072,737,1072,1427,1069,721,1078,1075,1425,735,1074,1075,1426,719,1074,1075,1426,13146,2189,1071,1059,1063,1072,720,1426,720,1092,1073,1426,1076,1070,720,1076,1076,1086,1078,1073,1074,1076,1423,1073,1073,733,1074,1424,1073,720,1077,1072,1425,735,1073,1074,1428,719,1076,1075,1425,13149,2191,1068,1067,1055,1073,722,1425,721,1092,1072,1424,1074,1070,722,1075,1074,1090,1076,1073,1076,1074,1427,1072,1069,735,1075,1426,1072,718,1076,1075,1426,733,1075,1077,1426,717,1078,1072,1428,13026,2187,1075,1067,1051,1074,720,1428,721,1091,1072,1428,1071,1074,721,1073,1075,1091,1073,1075,1073,1076,1426,1072,1070,735,1077,1422,1074,718,1076,1075,1425,733,1079,1072,1427,718,1076,1071,1424,13146,2191,1069,1060,1060,1072,722,1428,719,1088,1074,1427,1070,1073,720,1078,1072,1087,1076,1074,1078,1071,1426,1070,1071,735,1077,1426,1071,719,1074,1074,1426,733,1074,1074,1428,719,1075,1073,1426,13154,2184,1072,1064,1057,

Image:
[image]

And Two:
1087,1075,1423,1072,1071,720,1075,1071,1089,1075,1072,1074,1073,1424,1072,1067,734,1075,1424,1070,719,1074,1073,1427,732,1072,1073,1424,718,1074,1074,1423,13239,2191,1067,1053,1071,1067,721,1425,724,1085,1076,1427,1069,1074,720,1071,1076,1086,1076,1074,1078,1070,1424,1072,1071,734,1074,1424,1075,716,1074,1075,1425,734,1075,1074,1423,720,1074,1075,1422,13205,2189,1068,1060,1060,1073,719,1427,719,1091,1072,1427,1071,1070,720,1078,1071,1089,1074,1071,1074,1075,1425,1073,1069,734,1072,1426,1070,720,1077,1071,1425,736,1073,1076,1422,718,1076,1074,1423,13185,2191,1070,1072,1069,1070,721,1425,722,1086,1072,1426,1073,1069,720,1074,1078,1083,1074,1077,1070,1075,1422,1071,1071,734,1074,1427,1068,719,1073,1076,1422,732,1076,1072,1426,718,1073,1074,1425,13198,2186,1068,1057,1067,1071,723,1423,720,1089,1075,1424,1072,1071,721,1078,1071,1092,1071,1074,1074,1073,1424,1072,1071,735,1072,1426,1070,722,1073,1075,1423,733,1075,1075,1423,720,1072,1075,1425,13197,2189,1068,1053,1070,1070,720,1426,722,1089,1073,1425,1074,1069,721,1078,1070,1090,1073,1074,1074,1073,1425,1071,1074,730,1076,1424,1071,718,1074,1075,1424,736,1072,1075,1423,720,1074,1076,1423,13204,2186,1070,1063,1057,1073,721,1428,720,1089,1075,1427,1075,1068,721,1074,1074,1089,1076,1074,1073,1077,1423,1070,1076,729,1074,1427,1071,719,1076,1073,1426,730,1077,1073,1426,717,1076,1076,1424,13207,2190,1072,1050,1068,1073,720,1427,722,1088,1076,1423,1074,1070,720,1077,1074,1088,1074,1073,1076,1076,1423,1072,1070,734,1074,1424,1073,720,1072,1074,1428,733,1072,1076,1425,720,1073,1077,1424,13206,2187,1073,1050,1069,1074,720,1427,721,1088,1077,1425,1071,1069,722,1075,1075,1088,1075,1072,1074,1075,1425,1074,1068,733,1076,1424,1071,722,1071,1077,1425,733,1075,1075,1424,721,1075,1072,1426,13439,2191,1068,1050,1072,1071,724,1422,720,1089,1075,1423,1074,1072,718,1075,1077,1088,1074,1073,1074,1072,1427,1071,1068,734,1074,1423,1071,718,1077,1073,1426,733,1072,1075,1422,718,1076,1071,1426,13048,2187,1071,1069,1051,1071,722,1426,720,1093,1071,1425,1072,1070,721,1077,1072,1089,1074,1075,1074,1074,1424,1071,1071,735,1073,1426,1072,718,1078,1071,1431,730,1075,1074,1425,718,1078,1071,1426,13151,2186,1071,1070,1052,1071,722,1427,720,1091,1073,1428,1067,1071,721,1076,1075,1091,1071,1076,1074,1073,1427,1073,1070,732,1074,1427,1072,721,1074,1073,1424,734,1074,1075,1422,720,1074,1075,1425,13135,2185,1071,1072,1048,1072,719,1427,723,1083,1074,1424,1072,1073,722,1070,

Image:
[image]

joetriskaide, 2020-11-20:
Hi Martin,
I am struggling at the exact same issue; Just added six remote controlled blind to my home and now I am at the point to control them from a Pi or similar.
I am blocked with scanning already... I used a scanner on RPi (https://github.com/omerl13/raspberrypi-rf-dooya-controller) and a series on available and self-written programs on EPS8266, but could not get a consistent picture of the signal send.  I will continue with my research and post any outcomes here.

joetriskaide, 2020-11-26:
Hi @martin072 
I was successful in getting my blinds controlled from an ESP8266. I assumed that the remote I use would be more or less the same as the one you use. Looking at your oszi-image, I can see remarkable differences in the timing of the bits.
A look at the first few pulses (_1076,1426,13207,2188,1072,1062,1057,1073,719_,) unveils a few of them:
I assume the big gap of duration 13207µs is the gap between two repetitions of of the signal. A new one starts with 2188, 1072. My timings are different: ~4700/~1600 respectively. Another difference is the relation of the "1"-time vs. "0"-time. Your first data pulse (as I assume) is _2188 on_ and _1072 off_. The pulses I see in my remote are either 350 on/750 off or vice versa, i..e the pulses are about 2:1 in length while yours are about 1:1 in length. Same is true for  _1062 on_/_1057 off_) while _1073 on_/_719 off_ has another on/of time relation. 
I wrote an interim sketch for an ESP8266 to understand my Dooya remote and simply replay the timings which came out of https://test.sui.li/oszi/Sketchbook/SimpleRcScanner.ino. You can use the picture at https://test.sui.li/oszi/ to find repeating pulse patterns (in our case it would be good enough of look for timings >10000 for example)  and use this very simple sketch to send a copy of block fo the pattern out. (in your example above, I would try the numbers ,13207,2188,1072,1062,1057,1073,719,1428,720,1090,1075,1428,1073,1072,718,1077,1074,1090,1074,1074,1076,1071,1428,1070,1071,735,1076,1425,1070,720,1075,1076,1426,732,1074,1076,1426,719,1073,1074,1429,13200) 
If you are interested, I will polish my existing sketch a bit and upload it to a repository. Let me know.

martin072, 2020-11-27:
Hi joetriskaide,

Thanks for that. If I have some time this weekend, I will try the sketch.. I just realised the 2 uploads are from the same button, just two recordings as such. I am struggling a bit how to convert this to the protocol definition. 
The  2702 is a 16 Channel remote maybe that's why it differs so much. 

If there is anything I can supply to help, please let me know, I would love to get this thing working on the library (so I can use it with Tasmota)

joetriskaide, 2020-11-30:
Hi @martin072 ,
I managed to get the Dooya 1603 remote (is a 5-channel remote) working in Tasmota. What strikes me is that whatever I read for Dooya-remotes differs from what I see in you received timings. I am even not sure the pattern can be mapped at all to rc-switch.
What I understood from rc-switch is that the lib works on the assumption there are 2 pulses: one for a "1", one for a "0" and the pulses are usually a high state followed by a low state. the length of high vs. low state defines whether a "1" is transmitted or a "0", i.e. in my remote there is a "short, 340µs" high state followed by  a "long, 680µs" low state defining a "1" to be transmitted  Looking at the number you scanned, I can see more than two long/short and short/long patterns: There are patterns where the high state is as long as the low state (**1062,1057**), follow by a long-short pulse followed by another long-short pulse, where the "high" state is remarkable longer as in the previous pulse. 
I am new to rc-switch, but to my best knowledge, that is not the way the rc-switch lib works. They assume a fixed pattern for "1" and a fixed pattern for "0", not so much of a change like in your scan.
Having said that, there is the notion of triState which I haven't looked at because there wasn't any need for me and somebody with more knowledge on triStates chimes in help you with your remote.