sui77/rc-switch #375: Good library store rf433 frequency in a eprom and 868 mhz compatible

MaxESP, 2020-11-18:
I want to thank a lot the autor M Suat Özgür for your work that a very very good job .

I  have put your link in my website
http://misterdroid.ma/misterdroid/wigoHome.html

I have some question if you have time to discuss  I am trying to store the RF 433 mhz frequency in a rf receiver what is the best way to do this.

I all ready buy from china some 433 mhz  receiver module with a programmable function to store the RF frenquency of the remote controle in the receiver. i want to understand the process how it works.

2 point I am interested in 868 mhz because 433 mhz became very noisy is it possible to adapt your library to 868 mhz ook fsk module

best regards

Locke, 2020-11-18:
868 MHz usually uses a very different protocol, so I think it cannot be supported by this library.

However, I have heard about another library, maybe that could be something for you: https://github.com/SpaceTeddy/CC1101

Asteliks, 2020-11-18:
Great library if only it could work with 868 MHz that would really be perfect. I've got this blue HSM4-868 remote by hormann and have been trying to copy it with a rmf69hw for months now. Basically this is a PIC16F630 with a TH72032.2. Unfortunately CC1101 seems limited to 433MHz. Any suggestions?

MaxESP, 2020-11-19:
thanks Locke I have allready seen the link you have post  i am working on it now for 2 days .

MaxESP, 2020-11-19:
I think it would no be possible Asteliks  to read the frequency of your HSM4-868 remote because  rmf69hw don't take in charge 

ASK modulation has you give info about TH72032.2 tranceiver  better use cc1101 as Locke said you would have better chances

Asteliks, 2020-11-19:
the thing about cc1101 is that I can not seem to find the 868mhz version and the aliexpers offer seems strange. Dose it support all the frequencies at ones? RMF69HW was the only radio I was able to get at 868mhz

MaxESP, 2020-11-19:
https://fr.aliexpress.com/item/32803042083.html?spm=a2g0w.12010612.8148356.10.67f12e43gVxKsX
see this link

Asteliks, 2020-11-19:
nice thanks. Do you think this pinout will work with a breadboard?

MaxESP, 2020-11-19:
Sure depend of what mcu you have arduino esp ect......

MaxESP, 2020-11-19:
you would have to find the right lib in github

Asteliks, 2020-11-19:
ESP32 but I meant the white thing you put cc1101 on to prototype

Asteliks, 2020-11-19:
> you would have to find the right lib in github

probably I will just adjust the CC1101 lib for ESP32

MaxESP, 2020-11-19:
i think there is one allready done https://www.arduino.cc/reference/en/libraries/easy-esp-arduino-iot-with-cc1101/

Asteliks, 2020-11-19:
> i think there is one allready done https://www.arduino.cc/reference/en/libraries/easy-esp-arduino-iot-with-cc1101/

that should do, thanks