sui77/rc-switch #380: Decode RF Garage Door

dionisor, 2020-12-13:
Hello,
I'm trying to decode my remote with no luck.
I'm already tried:
- rtl_433:
> pi@raspbian:~ $ sudo rtl_433 -A -R 0
> rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
> Use -h for usage help and see https://triq.org/ for documentation.
> Trying conf file at "rtl_433.conf"...
> Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
> Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
> Trying conf file at "/etc/rtl_433/rtl_433.conf"...
> Disabling all device decoders.
> Registered 0 out of 175 device decoding protocols [ ]
> Found Fitipower FC0012 tuner
> Exact sample rate is: 250000.000414 Hz
> Sample rate set to 250000 S/s.
> Tuner gain set to Auto.
> Tuned to 433.920MHz.
> Allocating 15 zero-copy buffers
> 
> Detected OOK package    2020-12-12 22:46:29
> Analyzing pulses...
> Total count:  135,  width: 435.98 ms            (108996 S)
> Pulse width distribution:
>  [ 0] count:  105,  width: 1272 us [1264;1288]  ( 318 S)
>  [ 1] count:   30,  width: 2556 us [2552;2568]  ( 639 S)
> Gap width distribution:
>  [ 0] count:  100,  width: 1280 us [1268;1296]  ( 320 S)
>  [ 1] count:   30,  width: 2556 us [2548;2568]  ( 639 S)
>  [ 2] count:    4,  width: 5092 us [5088;5104]  (1273 S)
> Pulse period distribution:
>  [ 0] count:   90,  width: 2556 us [2548;2580]  ( 639 S)
>  [ 1] count:   25,  width: 3832 us [3824;3844]  ( 958 S)
>  [ 2] count:   15,  width: 5116 us [5108;5132]  (1279 S)
>  [ 3] count:    4,  width: 7648 us [7644;7656]  (1912 S)
> Pulse timing distribution:
>  [ 0] count:  205,  width: 1276 us [1264;1296]  ( 319 S)
>  [ 1] count:   60,  width: 2556 us [2548;2568]  ( 639 S)
>  [ 2] count:    4,  width: 5092 us [5088;5104]  (1273 S)
>  [ 3] count:    1,  width: 25684 us [25684;25684]       (6421 S)
> Level estimates [high, low]:   9722,    211
> RSSI: -2.3 dB SNR: 16.6 dB Noise: -18.9 dB
> Frequency offsets [F1, F2]:    9191,      0     (+35.1 kHz, +0.0 kHz)
> Guessing modulation: Pulse Width Modulation with multiple packets
> view at https://triq.org/pdv/#AAB025040404FC09FC13E4645480808080808080819191808080919080818080809080818080809255+AAB025040104FC09FC13E4645480808080808080819191808080919080818080809080818080809355
> Attempting demodulation... short_width: 1272, long_width: 2556, reset_limit: 5108, sync_width: 0
> Use a flex decoder with -X 'n=name,m=OOK_PWM,s=1272,l=2556,r=5108,g=2572,t=514,y=0'
> pulse_demod_pwm(): Analyzer Device
> bitbuffer:: Number of rows: 5
> [00] {27} ff 39 f7 c0 : 11111111 00111001 11110111 110
> [01] {27} ff 39 f7 c0 : 11111111 00111001 11110111 110
> [02] {27} ff 39 f7 c0 : 11111111 00111001 11110111 110
> [03] {27} ff 39 f7 c0 : 11111111 00111001 11110111 110
> [04] {27} ff 39 f7 c0 : 11111111 00111001 11110111 110

- SimpleRcScanner:

>  5... 4... 3... 2... 1...
> 192,470,653,172,358,226,170,354,4992,565,48,498,182,552,97,675,47,454,86,430,111,445,84,459,88,499,129,462,102,468,184,449,171,445,216,441,126,344,5854,592,97,837,89,517,169,512,174,519,165,519,31,515,128,467,169,488,166,486,169,464,115,486,179,423,125,452,134,633,89,454,177,306,197,645,183,358,5136,670,46,500,97,496,93,518,129,506,169,472,85,456,182,476,173,459,166,1082,181,490,111,612,132,500,175,453,128,419,176,346,216,42,367,225,164,375,178,362,182,361,210,414,29,549,131,477,127,467,166,605,171,507,107,488,168,471,168,456,94,497,175,458,114,510,179,509,179,420,187,432,171,452,177,422,209,456,175,394,177,439,187,365,273,620,129,326,199,38,92,27,279,203,103,324,3264,648,514,170,491,94,483,134,466,178,494,166,392,167,412,147,393,167,393,144,403,160,421,164,415,189,746,110,455,185,435,180,423,198,452,181,326,214,285,206,40,338,209,163,315,207,28,311,182,163,336,196,830,726,93,472,104,424,225,29,474,475,171,414,170,440,168,429,172,452,135,405,177,440,188,426,181,427,180,432,433,108,384,168,430,130,481,198,422,186,888,91,394,170,360,228,173,320,2590,523,97,513,100,516,118,459,137,436,134,441,185,447,176,484,179,441,126,476,29,465,167,437,166,496,84,441,96,437,192,413,181,405,181,408,190,401,182,397,209,29,412,108,44,110,115,314,212,164,325,9055,614,88,697,105,660,113,525,169,646,169,514,170,1079,36,525,131,379,208,110,355,190,53,22,385,193,41,364,191,131,30,436,224,110,401,130,524,86,478,182,511,113,460,184,566,32,446,172,484,174,509,169,471,103,428,172,514,49,471,187,497,170,429,186,425,166,440,94,1048,200,456,186,441,171,503,91,384,199,173,330,210,164,351,388,7741,622,43,478,109,518,144,475,149,492,151,454,109,442,249,714,92,416,167,412,163,372,219,192,196,31,513,132,357,194,92,308,1126,548,47,476,168,439,182,413,173,460,169,458,174,432,181,417,132,411,189,572,92,383,153,433,151,382,153,378,150,428,166,376,159,455,500,108,468,164,458,127,459,175,459,111,415,281,35,484,202,83,314,2751,924,90,725,131,473,406,127,424,156,494,204,
>  
>  5... 4... 3... 2... 1...
> 51,89,44,1046,151,675,178,901,255,47,980,290,90,1196,781,47,208,104,43,212,219,98,570,482,20239,783,256,143,315,178,123,315,2071,536,197,350,189,433,179,43,361,208,161,306,189,55,329,345,643,127,372,197,111,366,189,104,347,85,343,177,318,201,165,328,199,183,161,320,179,85,289,653,316,176,113,283,9091,646,131,433,165,452,177,567,135,386,170,416,189,401,187,385,180,426,180,363,191,30,379,185,344,166,359,212,112,334,8331,844,32,494,116,600,191,432,182,433,88,400,183,398,198,340,187,392,197,374,186,364,189,390,186,383,116,375,9558,703,116,561,88,628,160,541,165,486,116,474,165,529,181,440,175,446,164,455,182,350,217,171,365,126,367,197,44,343,179,316,210,161,363,4699,780,631,171,456,174,423,169,421,176,416,126,478,171,461,185,398,163,419,187,429,190,396,166,406,204,29,393,195,170,404,209,87,311,207,160,421,177,303,5858,639,89,544,106,484,50,470,130,499,129,512,130,468,128,441,166,464,166,474,180,436,188,456,198,667,134,351,2726,1035,91,550,94,539,34,449,89,538,175,580,795,93,477,171,465,168,466,188,450,181,433,167,463,165,466,129,423,164,418,172,487,97,422,186,504,88,464,144,360,149,266,1866,30,376,3776,993,31,623,115,880,47,485,113,477,85,477,168,488,179,433,34,426,166,480,169,460,175,471,168,518,35,505,98,427,33,467,166,339,203,44,343,175,330,272,38,371,169,334,193,45,330,3439,794,93,470,171,470,182,575,177,471,184,434,89,453,217,552,172,444,177,424,188,411,182,487,173,414,190,390,180,413,192,398,5520,634,89,532,88,509,114,486,165,512,95,464,125,489,165,498,115,462,105,506,166,534,165,478,187,580,171,462,148,506,103,505,182,426,167,370,209,316,220,42,315,227,124,339,156,348,2444,657,31,620,801,66,539,114,574,31,505,167,491,177,521,162,502,183,444,128,443,134,518,543,90,508,130,512,99,521,114,476,166,395,129,371,212,394,184,369,235,162,426,183,330,199,84,335,204,87,462,130,535,167,444,177,453,161,427,170,464,126,443,131,423,110,481,107,427,107,485,167,464,187,463,171,460,174,489,175,452,127,454,170,437,217,404,158,406,188,416,195,368,

- ReceiveDemo_Advanced:
Received 127 / 7bit Protocol: 0
data bits of pulse train duration: 17859
proposed protocol: { 1276, { 2, 4 }, { 1, 1 }, { 1, 1 }, false }
====
first level up
2548 5082 
1292 1271 1281 1257 1294 1261 1293 1272 1287 1250 1297 1259 1291 1254 
====
Received 15 / 7bit Protocol: 0
data bits of pulse train duration: 17891
proposed protocol: { 1278, { 2, 4 }, { 1, 1 }, { 1, 1 }, false }
====
first level up
2524 5055 
1291 1292 1251 1290 1271 1282 1283 1270 1285 1250 1296 1258 1291 1281 
====
Received 127 / 7bit Protocol: 0
data bits of pulse train duration: 17854
proposed protocol: { 1276, { 2, 4 }, { 1, 1 }, { 1, 1 }, false }
====
first level up
2530 5054 
1292 1275 1270 1264 1291 1255 1292 1272 1277 1257 1295 1257 1285 1272 
====
Received 95 / 7bit Protocol: 0
data bits of pulse train duration: 17869
proposed protocol: { 1277, { 2, 4 }, { 1, 1 }, { 1, 1 }, false }
====
first level up
2528 5055 
1288 1273 1273 1289 1270 1256 1292 1271 1282 1281 1269 1261 1292 1272 
====

- ProtocolAnalyzeDemo:

> Decimal: 127 (7Bit) Binary: 1111111 Tri-State: 111 PulseLength: 1277 microseconds Protocol: 0
> Raw data: 2526,5058,1287,1271,1306,1251,1294,1259,1299,1256,1289,1272,1276,1255,1319,
> 
> Decimal: 127 (7Bit) Binary: 1111111 Tri-State: 111 PulseLength: 1276 microseconds Protocol: 0
> Raw data: 2525,5058,1286,1273,1297,1297,1259,1258,1289,1271,1304,1252,1292,1264,1291,
> 
> Decimal: 127 (7Bit) Binary: 1111111 Tri-State: 111 PulseLength: 1276 microseconds Protocol: 0
> Raw data: 2504,5044,1337,1224,1279,1276,1296,1262,1289,1251,1286,1274,1297,1256,1296,

When I check my signal with rtl_433 the result is different.
I tesdted:
> mySwitch.setProtocol({ 1272, { 2, 4 }, { 1, 1 }, { 2, 2 }, false })
and
> mySwitch.setProtocol({ 1272, { 1, 4 }, { 1, 1 }, { 2, 2 }, false })

Can anyone help me to find protocolo and code to send.

Any help would be appreciated.

cgmckeever, 2020-12-13:
Where is this output coming from ? My advanced demo gives me different output
```
Received 127 / 7bit Protocol: 0
data bits of pulse train duration: 17854
proposed protocol: { 1276, { 2, 4 }, { 1, 1 }, { 1, 1 }, false }
```

dionisor, 2020-12-14:
> Where is this output coming from ? My advanced demo gives me different output
> 
> ```
> Received 127 / 7bit Protocol: 0
> data bits of pulse train duration: 17854
> proposed protocol: { 1276, { 2, 4 }, { 1, 1 }, { 1, 1 }, false }
> ```

What do you refer to?
I'm using [ReceiveDemo_Advanced](https://github.com/sui77/rc-switch/tree/master/examples/ReceiveDemo_Advanced)
Could you share your output?

cgmckeever, 2020-12-14:
This is my output from ReceiveDemo_Advanced

```
Decimal: 7 (3Bit) Binary: 111 Tri-State: 1 PulseLength: 37 microseconds Protocol: 0
Raw data: 2256,484,100,52,116,36,124,
```