sui77/rc-switch #383: Add support for DiO

guyou, 2020-12-29:
I'm trying to add support for my remote: a DiO.

A sample recorded with SimpleRCscanner:
```
120,116,192,740,352,1124,224,188,120,76,548,440,400,356,116,1148,260,408,568,752,464,1528,296,120,140,280,332,1080,92,612,392,92,600,452,1184,400,108,552,112,480,1200,468,128,532,1120,660,992,464,168,440,112,416,620,72,540,568,976,100,52,508,76,440,1228,472,116,432,1100,64,76,456,152,448,1192,388,200,476,104,384,1296,416,168,716,952,396,1264,428,192,392,1256,420,156,420,152,436,1256,412,1240,432,156,432,1252,440,156,388,1248,420,176,412,180,432,1228,384,200,416,1256,472,1180,412,212,428,1224,408,180,412,164,504,1160,388,208,468,1196,424,1244,400,200,368,216,340,1332,368,216,364,1308,344,236,352,1316,376,204,364,1308,368,10796,356,2772,368,216,364,1304,340,1320,356,236,368,1288,364,228,372,204,376,1304,372,1292,368,216,344,236,388,1276,388,200,376,1304,348,1308,364,244,352,236,344,1320,344,1316,364,228,348,1312,344,248,360,1304,348,236,356,1308,360,232,356,220,372,1300,372,208,372,1304,364,1300,356,244,356,1308,352,236,360,224,352,1328,352,1304,344,244,356,1304,364,228,348,1316,360,224,364,216,364,1316,364,212,360,1316,364,1292,368,232,364,1304,360,232,352,228,364,1316,344,232,348,1320,360,1304,356,232,364,220,360,1308,360,224,356,1316,356,220,360,1312,364,216,368,1300,360,10808,380,2744,392,188,392,1284,396,1260,400,192,392,1276,388,196,380,204,380,1292,380,1280,388,204,388,196,380,1292,380,200,384,1284,380,1284,384,216,388,200,384,1284,388,1272,388,204,384,1280,392,192,384,1284,384,208,380,1284,380,208,380,200,384,1288,384,204,376,1292,376,1284,388,216,380,1280,384,208,384,196,384,1288,388,1276,388,204,384,1276,392,200,388,1276,376,212,380,204,376,1296,384,200,376,1296,376,1284,384,220,380,1280,380,208,388,196,384,1288,380,204,384,1284,388,1276,384,208,384,196,380,1288,400,180,400,1272,400,180,404,1268,400,180,404,1264,396,10768,428,2704,432,148,432,1236,424,1240,428,164,428,1232,428,164,428,152,424,1252,424,1236,424,164,420,164,420,1252,424,156,424,1252,416,1244,420,184,416,164,416,1260,412,1248,424,168,420,1240,416,176,420,1244,412,176,420,1244,420,172,412,168,412,1260,416,164,412,1264,412,1248,416,188,416,1248,412,176,416,164,416,1260,416,1244,412,180,412,1252,412,180,
```

I identified the following sequence:
```
10796,356,2772,368,216,364,1304,340,1320,356,236,368,1288,364,228,372,204,376,1304,372,1292,368,216,344,236,388,1276,388,200,376,1304,348,1308,364,244,352,236,344,1320,344,1316,364,228,348,1312,344,248,360,1304,348,236,356,1308,360,232,356,220,372,1300,372,208,372,1304,364,1300,356,244,356,1308,352,236,360,224,352,1328,352,1304,344,244,356,1304,364,228,348,1316,360,224,364,216,364,1316,364,212,360,1316,364,1292,368,232,364,1304,360,232,352,228,364,1316,344,232,348,1320,360,1304,356,232,364,220,360,1308,360,224,356,1316,356,220,360,1312,364,216,368,1300,360
```

I think the protocol is based on a common separator (~360), a long value (~1300) and a short value (~200). But I think the protocol uses a specific synch with two values : 10796 and 2772.

Is it possible to use RCswtich with such protocol?

wibimaster, 2021-01-05:
@guyou What's your DI-O remote ?
Cannot recognize my remote too, ref 54794, type LYCT-819
I'm coming from a RPi (not Arduino) and search for a way to get protocol and value emitted...

Thanks !

guyou, 2021-01-24:
> @guyou What's your DI-O remote ?
> Cannot recognize my remote too, ref 54794, type LYCT-819
> I'm coming from a RPi (not Arduino) and search for a way to get protocol and value emitted...

My remote has ref 54760 and the whole packaging is referenced as 54797.
In order to hae something working, I had to switch to another solution than RC-Switch and adapt the timings. The other solution is https://github.com/guyou/DiOremote and the changes are https://github.com/guyou/DiOremote/compare/ref-54760
The way I found these values is described in my blog (french): http://nathguil.free.fr/blog/posts/controle-modules-dio-433mhz/

Keep us informed of your success.

1technophile, 2021-01-24:
This library support kaku protocol for info : 
https://github.com/1technophile/NewRemoteSwitch

guyou, 2021-01-24:
> This library support kaku protocol for info :
> https://github.com/1technophile/NewRemoteSwitch

@1technophile do you think DiO (Chacon) is using Kaku protocol? Rotation of timings seems similar. But the base timing seems quite different. Here are the values collected:
```
Timings 3× 10790.7 ±20.0 µs     173× 383.8 ±46.0 µs     3× 2740.0 ±34.0 µs      72× 209.8 ±38.0 µs      84× 1283.2 ±48.0 µs     12× 161.7 ±10.0 µs
```

I also noticed your version of RC-Switch, supporting much more complex protocols: https://github.com/1technophile/rc-switch

1technophile, 2021-01-24:
> @1technophile do you think DiO (Chacon) is using Kaku protocol? 

At least I have a DIO Chacon remote control and switch using Kaku and compatible with newremoteswitch.
https://compatible.openmqttgateway.com/index.php/product/433mhz-dio-chacon-outlet/