sui77/rc-switch #390: Another roller shutter protocol?

caitken-com, 2021-01-05:
I was unable to capture my (Australian) window roller shutter's remote codes with rc-switch lib. I did manage to successfully capture and re-transmit the codes using [Tasmota on a Sonoff RF bridge](https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/). But Tasmota has a wifi connection conflict with my current modem/router on restart, so I'm hoping to use rc-switch lib on an Uno with ethernet shield instead.

Is it possible for rc-switch to support the follow protocol?

### Tasmota's rfraw results for UP command:

> 32 05 08 07F8 0FB4 01EA 03D4 2332 4808092A3A2A2A3B2A2A2A2B3A2A2A3A2A2A2A3A2B2B3B2A3B3A2A2B3A2B2A3A2A3B2A3A3B2A

|Hex code|Description|
|--|--|
|32|Length, in bytes|
|05|Number of buckets|
|08|Repetition count|
|07F8|Bucket 0 length: 2040µs|
|0FB4|Bucket 1 length: 4020µs|
|01EA|Bucket 2 length: 490µs|
|03D4|Bucket 3 length: 980µs|
|2332|Bucket 4 length: 9010µs|
|480809...|RF data to send (Each nibble represents bucket number buckets and high/low marking)|

### Tasmota's rfraw results for DOWN command:

> 32 05 08 07EE 0FB4 01EA 03DE 233C 4808092A2A2A2A3B2A2A2A2B3A2A2A3A2A2A2A3A2B2B3B2A3B3A2A2B3A2B2A3A2B3B2A3A3B2A

|Hex code|Description|
|--|--|
|32|Length, in bytes|
|05|Number of buckets|
|08|Repetition count|
|07EE|Bucket 0 length: 2030µs|
|0FB4|Bucket 1 length: 4020µs|
|01EA|Bucket 2 length: 490µs|
|03DE|Bucket 3 length: 990µs|
|233C|Bucket 4 length: 9020µs|
|480809...|RF data to send (Each nibble represents bucket number buckets and high/low marking)|

I haven't seen enough documentation properly explaining how to use custom protocols, such as the given example does what exactly?
`static const RCSwitch::Protocol customprotocol = { 360, {  1, 23 }, {  1,  2 }, { 2,  1 }, false };
mySwitch.setProtocol(customprotocol);`

avbdr, 2022-07-21:
@caitken-com were you able to solve this? I have the same issue

caitken-com, 2022-07-27:
@avbdr Hi, no I haven’t. I moved my routing and access points over to Unifi gear. So now the Sonoff RF bridge (running Tasmota) works flawlessly.