sui77/rc-switch #394: Regular crashes on ESP32

DriekdeGadgetfreak, 2021-01-19:
According to the expressif documentation:

If the ISR is placed into IRAM, all constant data used by the ISR and functions called from ISR (including, but not limited to, const char arrays), must be placed into DRAM using DRAM_ATTR.

So:

static const RCSwitch::Protocol proto[]
and
const unsigned int RCSwitch::nSeparationLimit = 4300;

should be changed to:

static const DRAM_ATTR RCSwitch::Protocol proto[]
and
const DRAM_ATTR unsigned int RCSwitch::nSeparationLimit = 4300;

1technophile, 2021-01-19:
For the protocol you have it already into the master below:
https://github.com/sui77/rc-switch/commit/ebe91719936265540adcb9766f6ed6e09befe48f

DriekdeGadgetfreak, 2021-01-20:
AH, id not see that!
Thanx.