sui77/rc-switch #397: Doesn't work on DOIT ESP32 Devkit V1

mcpacific, 2021-01-30:
I've loaded up ReceiveDemo_Advanced in my Arduino IDE and uploaded it to my DOIT ESP32 DevKit V1 board, but I can't seem to get it to work. I thought it would be best to use GPIO 4 or GDIO 23 but neither works. In fact, any interrupt I use doesn't seem to have an impact on the mySwitch.available() statement in the loop.

```
void setup() {
  Serial.begin(115200);
  Serial.println();
  mySwitch.enableReceive(4);
}
```

I included a serial output in the loop if mySwitch ever becomes available, but it does not:

```
void loop() {
  if (mySwitch.available()) {
    Serial.println("mySwitch is available.");
    output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
    mySwitch.resetAvailable();
  }
}
```

Pressing buttons on the remote produces no output.

1technophile, 2021-01-30:
Try with 27 or 26 for the receiver on ESP32

mcpacific, 2021-01-31:
I tried both 27 and 26, still nothing. I even tried different RF receivers... it's like it never becomes available.

1technophile, 2021-01-31:
Are you sure your emitter is emitting a compatible signal?

If you are not sure, you may build a setup with rcswitch and a simple [sendDemo](https://github.com/sui77/rc-switch/tree/master/examples/SendDemo) example.

mcpacific, 2021-02-02:
1technophile, I assume you mean upload sendDemo to another ESP32 and see if the original ESP32 receives it, is that correct?

1technophile, 2021-02-02:
> 1technophile, I assume you mean upload sendDemo to another ESP32 and see if the original ESP32 receives it, is that correct?

Yes this is correct.

Could you post your sketch and pictures of your emitter and receiver setup?

mcpacific, 2021-02-05:
I was finally able to get this to work. The problem was, there were multiple bad transmitters and receivers in each package from BOTH vendors.

closed 2021-02-06 by mcpacific