sui77/rc-switch #413: Unable to receive using ReceiveDemo_Simple  from SendDemo

paulgeanta, 2021-03-28:
Maybe I did not understand the purpose of the library but I have spent 2 days trying to make this work, and I'm really sorry if the question is just a question and probably not an issue.

At this point I try to send using the the SendDemo example and receive using the ReceiveDemo_simple example (I guess there is no need to add the code). I prefer to be able to do this before I try anything else involving real remote controls

I know that the boards are good since I was able to make them communicate using the RadioHat library.

I use 2 [https://www.amazon.com/gp/product/B08BJNV1J3/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1](url) and [https://www.amazon.com/gp/product/B086ZL8W1W/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1](url).

I also tried with [https://www.amazon.com/gp/product/B07DW9K3Q7/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1](url) even though I would guess this is the wrong frequency.

Can you please give me some pointers. Anything would be appreciated (be it just ... those boards are no good) so I stop struggeling.

Thank you,
Paul

1technophile, 2021-03-28:
Could you post a picture of your setup?

paulgeanta, 2021-03-28:
Here they are.

Thank you,
Paul

[image]
[image]
[image]

1technophile, 2021-03-28:
Seems ok, did you checked with a multimeter that the modules are powered correctly (5V) and that there is connection between the modules data pins and the micro pins?

paulgeanta, 2021-03-28:
Not with a multimeter, but using Radiohead library example (ask_transmitter and ask_reciver) I get good results using the exact same pins.

[image]

I assume that in this case the voltage should be ok. Or am I wrong?

Thank you,
Paul

1technophile, 2021-03-28:
Yes it should. Are you using the examples sketch without any modifications ?

paulgeanta, 2021-03-28:
The only modification is how I instantiate, in order to change the default pins

RH_ASK driver(2000,2,10); instead of RH_ASK driver;

Thank you,
Paul

1technophile, 2021-03-28:
I mean the RCSwitch ones?

paulgeanta, 2021-03-28:
Oh sorry. I made absolutely no changes to the RCSwitch examples. 

Trying with SendDemo and ReceiveDemo_Simple now. I also tried ReceiveDemo_advanced and it did not work either

Thank you,
Paul

pierzogad, 2021-05-17:
It's a bit late but maybe it would help others. You need to modify SendDemo_... example to enable transmit on pin 10 - i.e. add 

`mySwitch.enableTransmit(10);` to setup() 

Without it pin 10 on receiving module is left in INPUT mode which causes transmitter to generate constant "noise" that prevents communication - or at least this is the issue I had.

petsch9, 2022-03-15:
Same problem here with RC-switch i do not resive any signal.
I tryed on ESP32 and NANO but no result.

mySwitch.enableTransmit(10); dit not change anything

newremoteswitch libary works but this only works with one protocol

someone any ideas?