sui77/rc-switch #417: Wifi Disconnects Randomly - Arduino Nano(WifiEsp Library), ESP8266-01, MQTT(PubSubClient)

anil8anil, 2021-04-23:
Hi everyone,
I have been facing an issue with my wifi connection. Actually i have tried so many different approaches until i could find the right spot to solve. I will describe shortly;

1. My setup:
- My aim is to receive RF commands through an RF receiver and publish the to MQTT.
- I am using ESP8266-01 (AT firmware v1.1.1) connected to a Ardunio Nano. I use [WifiEsp ](https://github.com/bportaluri/WiFiEsp) library to connect to wifi. I use SoftwareSerial library to communicate with ESP.
- I also connect an RF Receiver and connect it to interrupt0 (D2 pin) of Nano.
- I don't prefer to use ESP as controller and directly connect RF receiver to it. Because i couldnt solve some problems on delays and fails about interrupts on ESP: Issue #284 

2. My problem:
- If I enable receive [enableReceive(rfReceiverPin)] in my code (even if i dont do anything in loop about rf receiving), I face too many disconnects from wifi, connects sometimes but disconnects after a while maybe 5 seconds. I cannot attain a stable connection. Reconneting attempts take long times also.
- If I don't enable receive, There is no problem. I attain stable connection to wifi and also to mqtt broker. I face disconnects sometimes but once in 2-5 minutes and it reconnects in just a second. It is not a problem for me.

I think there is a problem about using interrupts. I am not an expert but i guess wifiesp library or pubsubclient library use interrupts somehow and when i try to use interrupt0 pin for RF receiver, problems occur.

Do you have any idea about the problem? Thanks in advance. Here are my codes:

```
#include <WiFiEspClient.h>
#include <WiFiEsp.h>
#include <WiFiEspUdp.h>
#include <PubSubClient.h>
#include "SoftwareSerial.h"
#include <RCSwitch.h>                     
#include <SimpleTimer.h>                  

#define WIFI_AP "mySDID"
#define WIFI_PASSWORD "myPASSWORD"

char mqttServer[] = "192.168.1.23";
int status = WL_IDLE_STATUS;

WiFiEspClient espClient;                  // Initialize the Ethernet client object
PubSubClient client(espClient);
SoftwareSerial soft(10, 11);              // RX, TX

RCSwitch rfSwitch = RCSwitch();           
static const uint8_t rfReceiverPin = 0;   
SimpleTimer timer;                        
int timerId;                              

void setup() {
  Serial.begin(9600);
  InitWiFi();
  client.setServer(mqttServer, 1883);
  rfSwitch.enableReceive(rfReceiverPin);  //MY PROBLEM IS OCCURS WHEN I USE THIS LINE
  timerId = timer.setInterval(5000L, timerFunction);
}

void loop() {
  checkWifiConnection();
  checkMQTTConnection();
  //I DONT EVEN DO ANYTHING IN LOOP ABOUT RF RECEIVING. NORMALLY I USE MY RFCONNECTION FUNCTION HERE.
  timer.run();
}

void timerFunction(){                 
  timer.disable(timerId);
  if(client.publish("rfbridge", "test")){
      Serial.print("MQTT Published - Topic:");
      Serial.print("rfbridge");
      Serial.print(" - Payload:");
      Serial.println("test");
  } else{
      Serial.print("MQTT publish FAILED - Topic:");
      Serial.print("rfbridge");
      Serial.print(" - Payload:");
      Serial.println("test");
  }
  timer.enable(timerId);
}

void checkWifiConnection(){
  timer.disable(timerId);
  if ( status != WL_CONNECTED) {
    while ( status != WL_CONNECTED) {
      Serial.print("Attempting to connect to WPA SSID: ");
      Serial.println(WIFI_AP);
      status = WiFi.begin(WIFI_AP, WIFI_PASSWORD);          // Connect to WPA/WPA2 network
      delay(500);
    }
    Serial.println("Connected to AP");
  }
  timer.enable(timerId);
}

void checkMQTTConnection(){
  timer.disable(timerId);
  if ( !client.connected() ) {
    reconnect();
  }
  client.loop();
  timer.enable(timerId);
}

void InitWiFi(){ 
  soft.begin(9600);                                         // initialize serial for ESP module
  WiFi.init(&soft);                                         // initialize ESP module
  
  if (WiFi.status() == WL_NO_SHIELD) {                      // check for the presence of the shield 
    Serial.println("WiFi shield not present");  
    while (true);                                           // don't continue
  }
  
  Serial.println("Connecting to AP ...");                   // attempt to connect to WiFi network  
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(WIFI_AP);
    // Connect to WPA/WPA2 network
    status = WiFi.begin(WIFI_AP, WIFI_PASSWORD);
    delay(500);
  }
  Serial.println("Connected to AP");
}

void reconnect() { 
  while (!client.connected()) {                                         // Loop until we're reconnected
    Serial.print("Connecting to MQTT Server ..."); 
    if (client.connect("RfBridge", "mqttuser", "abali08abali08")) {     // Attempt to connect (clientId, username, password)
      Serial.println("[DONE]");
    } else {
      Serial.print("[FAILED] [ rc = ");
      Serial.print(client.state());
      Serial.println(" : retrying in 5 seconds]");    
      delay(5000);                                                      // Wait 5 seconds before retrying
    }
  }
}
```

1technophile, 2021-04-23:
You may take a look at [OpenMQTTGateway ](https://docs.openmqttgateway.com/) and in particular [ZgatewayRF.ino](https://github.com/1technophile/OpenMQTTGateway/blob/development/main/ZgatewayRF.ino). It does exactly what you are trying to do (and more).

All the network handling is into main.ino

anil8anil, 2021-04-23:
> You may take a look at [OpenMQTTGateway ](https://docs.openmqttgateway.com/) and in particular [ZgatewayRF.ino](https://github.com/1technophile/OpenMQTTGateway/blob/development/main/ZgatewayRF.ino). It does what you are trying to do (and more).
> You will not need the nano for your use case just the ESP + 433mhz receiver would work.
> 
> All the network handling is into main.ino

Thanks @1technophile for your reply, i saw both but didnt try to use them yet. I tried to use Tasmota on my NodeMCU (ESP controller) with an RF receiver attached, i worked well actually, but there was some delay with tasmota too. When i pressed transmitter's button for a short time (for example less then a second), tasmota didn't receive the RF message. It received only if i pressed trasmitter's button for a second or more.

After that try with tasmota, i changed my setup to Nano+ESP. But i will give it a try with  OpenMQTTGateway and ZgatewayRF.ino too, to see their performance for my use case... But before that i need to find a solution or an answer about the issue i faced with Nano+ESP...

anil8anil, 2021-04-25:
Update:
- I tried "[ArduinoMqttClient](https://github.com/arduino-libraries/ArduinoMqttClient)" library for MQTT also. I faced similar problems. But after a while i've figured out that "not repeating the code block below" solved my problems. I would recommend to try this technique who face similar problem. I repeat code below (in a function with timer) only once in 5 seconds:
```
if ( !client.connected() ) {
    reconnect();
  }
client.loop();
```
My wifi and mqtt connections are mostly stable now. But mostly. I didnt monitor yet how frequent the drops are. 

- Also after @1technophile 's advice, i tried OpenMQTTGateway on Nodemcu. That was much more easy and stable. I guess i will go on with OpenMQTTGateway. Thanks @1technophile

closed 2021-04-25 by anil8anil