sui77/rc-switch #421: Broken transmit protocol & pulse length

ppartarr, 2021-05-22:
## Equipement
I'm using an FS1000A transmitter & XY-MK-5V receiver to control the brennenstuhl RC CE1 4001 set

## Problem
Sniffing the controller's signal whilst repeatedly pressing the ON button for the A channel will cycle through these 4 messages:

```bash
Decimal: 11845292 (24Bit) Binary: 101101001011111010101100 Tri-State: not applicable PulseLength: 100 microseconds Protocol: 3
Raw data: 7128,952,568,444,1072,944,564,952,564,444,1072,948,564,444,1072,440,1072,948,572,436,1072,944,572,944,572,944,568,944,572,944,572,436,1072,948,572,436,1072,948,568,440,1076,940,572,944,568,444,1068,444,1080,

Decimal: 11538540 (24Bit) Binary: 101100000001000001101100 Tri-State: not applicable PulseLength: 100 microseconds Protocol: 3
Raw data: 7148,948,564,448,1084,948,572,948,572,440,1068,448,1072,444,1068,448,1072,444,1068,448,1068,448,1068,952,572,440,1068,448,1072,444,1072,444,1072,444,1064,956,572,944,568,448,1072,948,564,952,568,444,1076,440,168,

Decimal: 12557084 (24Bit) Binary: 101111111001101100011100 Tri-State: not applicable PulseLength: 100 microseconds Protocol: 3
Raw data: 7144,940,572,440,1076,948,568,944,572,940,572,944,576,936,580,936,576,940,576,432,1076,440,1076,940,572,944,568,440,1076,944,572,944,568,440,1076,436,1076,436,1084,936,572,940,576,940,580,432,1088,424,1080,

Decimal: 12036188 (24Bit) Binary: 101101111010100001011100 Tri-State: not applicable PulseLength: 100 microseconds Protocol: 3
Raw data: 7136,948,572,440,1076,948,568,944,572,436,1072,948,568,944,568,948,572,940,576,436,1076,940,572,436,1080,940,572,440,1076,440,1072,440,1072,440,1072,948,564,444,1080,936,580,936,572,940,576,436,1076,440,1076,
```

I am therefore using the arduino sketch below to activate the switch with the sender

```bash
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
    Serial.begin(9600);
    mySwitch.enableTransmit(12);
    mySwitch.setProtocol(3);
    mySwitch.setPulseLength(100);
    Serial.println("init rcswitch sender");
}

void loop() {
    transmit(12036188);
    transmit(11845292);
    transmit(12557084);
    transmit(11538540);

    delay(5000);
}

void transmit(unsigned long code) {
  mySwitch.send(code, 24);
  delay(1000);
}
```

And this results in the following messages:
```bash
Decimal: 11845292 (24Bit) Binary: 101101001011111010101100 Tri-State: not applicable PulseLength: 512 microseconds Protocol: 5
Raw data: 7188,892,640,380,1152,892,636,884,644,392,1160,860,648,376,1156,376,1164,864,740,284,1136,896,632,892,628,892,652,872,656,864,668,360,1140,892,628,388,1132,892,628,392,1220,804,732,784,668,352,1160,356,1260,

Decimal: 12557084 (24Bit) Binary: 101111111001101100011100 Tri-State: not applicable PulseLength: 512 microseconds Protocol: 5
Raw data: 7188,896,636,388,1140,872,664,856,680,856,668,856,668,864,656,868,648,876,648,368,1168,360,1164,860,668,856,676,344,1180,844,668,860,652,364,1248,276,1160,364,1160,860,652,868,664,852,652,364,1140,384,1236,

Decimal: 11538540 (24Bit) Binary: 101100000001000001101100 Tri-State: not applicable PulseLength: 519 microseconds Protocol: 5
Raw data: 7176,896,640,392,1136,888,636,892,636,392,1140,388,1140,384,1140,388,1136,356,1160,372,1152,372,1136,896,628,392,1136,392,1128,396,1132,380,1136,388,1148,876,648,876,644,376,1220,804,616,888,632,392,1132,392,1136,

Decimal: 12036188 (24Bit) Binary: 101101111010100001011100 Tri-State: not applicable PulseLength: 519 microseconds Protocol: 5
Raw data: 7284,784,740,288,1152,792,648,880,648,380,1152,880,644,876,656,872,656,868,656,368,1156,876,732,292,1152,872,728,296,1232,288,1152,368,1152,372,1156,868,732,288,1152,872,728,796,644,876,652,368,1152,368,1152,
```

As you can see, the received pulse length is 512 instead of the 100, and the protocol is 5 instead of 3. The received values show that the sender is using the wrong protocol and the wrong pulse length even though they have been correctly defined. The switch is therefore not turning on.

Any ideas on why the library isn't working as intended?

ppartarr, 2021-05-22:
Okay after a long debugging session, it seems like `mySwitch.setProtocol(3);` is simply the wrong protocol to use for the brennenstuhl RC CE1 4001 even though the captured protocol is clearly 3... 

This user on a german arduino forum [1] is using the same model as I am and managed to get it working by setting the protocol to 4. I can confirm that this also worked on my end.

Although I'm glad my issue is solved I'm still puzzled as to why the receiver was showing protocol 3 when sniffing the controller's messages and why replaying the same message would not work...
 
[1] https://forum.arduino.cc/t/brennenstuhl-rc-ce1-4001-433mhz-kein-vorankommen/493589

iamnotsafe, 2021-08-23:
I have a very similar and weird issue; the pulse length that I set (176) is not what is transmitted (352). Everything else looks to be transmitted to what I set it to do (Decimal code, Binary code, Protocol number) except for the Raw data that looks different too.

Why is this happening?

litamin, 2021-08-23:
I have a very similar and weird issue; the pulse length that I set (176) is not what is transmitted (352). Everything else looks to be transmitted to what I set it to do (Decimal code, Binary code, Protocol number) except for the Raw data that looks different too.

Why is this happening?