sui77/rc-switch #423: no code shows when remote is activated

schaffertown, 2021-05-25:
I am using a nano and a 315 mhz receiver with 5 different remotes. I have used simple and advanced receive demos with no success. I have an antenna soldered on the board and have it wired to pin d2, power and ground. I bought a 5 pack of boards none have worked. What am I missing?

pierzogad, 2021-05-28:
Are you sure that:
* remotes are sending on same 315 mhz frequency?
* no other devices are connected to it?

Try pulling my branch [https://github.com/pierzogad/rc-switch/tree/receiving-patch/examples/ReceiveDemo_Advanced](https://github.com/pierzogad/rc-switch/tree/receiving-patch/examples/ReceiveDemo_Advanced) this will allow you to see nonstandard timing output.

schaffertown, 2021-05-28:
Thank you for the response. I tried the code you have. I get error codes on these 2 lines.
  mySwitch.setReceiveUsingProtocolTiming(false);
  
  mySwitch.setReceiveUnknownProtocol(true);

C:\Users\Michael\Documents\Arduino\sketch_may24a\sketch_may24a.ino: In function 'void setup()':
sketch_may24a:20:12: error: 'class RCSwitch' has no member named 'setReceiveUsingProtocolTiming'; did you mean 'getReceivedProtocol'?
   mySwitch.setReceiveUsingProtocolTiming(false);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            getReceivedProtocol
sketch_may24a:22:12: error: 'class RCSwitch' has no member named 'setReceiveUnknownProtocol'; did you mean 'getReceivedProtocol'?
   mySwitch.setReceiveUnknownProtocol(true);
            ^~~~~~~~~~~~~~~~~~~~~~~~~
            getReceivedProtocol
C:\Users\Michael\Documents\Arduino\sketch_may24a\sketch_may24a.ino: In function 'void loop()':
sketch_may24a:27:5: error: 'output' was not declared in this scope
     output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
     ^~~~~~
exit status 1
'class RCSwitch' has no member named 'setReceiveUsingProtocolTiming'; did you mean 'getReceivedProtocol'?

I am not sure what it is asking me to do. the original code loads fine. Is there a chance the rcswitch library did not install propperly from the library manager?

pierzogad, 2021-05-29:
I was refering to my modified clone of original code https://github.com/pierzogad/rc-switch/tree/receiving-patch 
Unfortunatelly these changes are not part of "official" deployment yet. 

To use this code you need to clone my repository into arduino's library directory using git.

Quick and dirty way to do it would also be:
1. Find `RCSwitch.cpp` and `RCSwitch.h` files on your computer (My Documents -> Arduino -> libraries -> rc-switch)
2. replace these files with https://github.com/pierzogad/rc-switch/blob/receiving-patch/RCSwitch.cpp and https://github.com/pierzogad/rc-switch/blob/receiving-patch/RCSwitch.h respectively.

Once you've finished your tests you may re-install library in library manager to revert to original.

schaffertown, 2021-05-30:
Thank you. I was able to get all the signals to show up in serial monitor. The range is terrible. Without an antenna soldered on 1 inch range, with an antenna 18 inches range. The voltage from my computer's usb is 4.3 volts. If that is as good as it gets I will need to buy a Receiver made for 433 or 315 openers.