sui77/rc-switch #430: Does not work with remote NICE for stores

aterzaghi, 2021-08-08:
Hello,
thanks for you your wonderful library.
I was looking to grab and reproduce the commands of this remote in order to automate them in an Arduino/Pi/Nodered infrastructure.

https://www.habitat-automatisme.com/nice-plano-era-p1.html

I loaded your example ReceiveDemo_Advanced with no changes and connected a normal 433MHy device to pin #2 

https://www.banggood.com/Geekcreit-433Mhz-RF-Decoder-Transmitter-With-Receiver-Module-Kit-For-ARM-MCU-Wireless-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-74102.html?ab_toppage_rr1=2&p=MA240439985285201910&cur_warehouse=CN

your library do not detect the remote... maybe there is any hope, just tweaking something?
many thanks
Andrea

pierzogad, 2021-08-13:
Did you start with connecting receiver only or both receiver and transmitter? 
If transmitter is present you need to add 
` mySwitch.enableTransmit(<transmitter pin>);`

aterzaghi, 2021-08-14:
Dear Pierzogad,

thank your for your answer.
I followed exactly what described in the page, just copying the code that compiled perfectly and  attaching receiver ONLY.

thank you for any idea.