sui77/rc-switch #435: Receiving two different values when pressing ON button

Pozee, 2021-09-21:
Pressing ON 1st time:
Decimal: 3923901139 (32Bit) Binary: 11101001111000011111101011010011 Tri-State: not applicable PulseLength: 702 microseconds Protocol: 2

Decimal: 3923901139 (32Bit) Binary: 11101001111000011111101011010011 Tri-State: not applicable PulseLength: 720 microseconds Protocol: 2

Pressing ON 2nd time:
Decimal: 854420883 (32Bit) Binary: 00110010111011010110110110010011 Tri-State: not applicable PulseLength: 702 microseconds Protocol: 2

Decimal: 854420883 (32Bit) Binary: 00110010111011010110110110010011 Tri-State: not applicable PulseLength: 720 microseconds Protocol: 2

So when i press ON on my remote it alternates between the response above. Im not 100% sure how i should go about transmitting these. Iäve tried the SendDemo but no matter what config i try my outlet won't turn on. I have tried changing Protocol from 1-6, tried a couple pulse lengths and tried sending binary & decimal without any success. Anyone got any tips?

aronlanza, 2021-12-08:
Hello I have the same problem as you, what power sockets do you use? 
Here is my sniffing result:

ON
15:33:25.381 -> Decimal: 1466527123 (32Bit) Binary: 01010111011010010110110110010011 Tri-State: not applicable PulseLength: 704 microseconds Protocol: 2
15:33:25.518 -> Raw data: 7048,472,1524,1476,524,476,1524,1472,528,472,1532,1464,528,1472,528,1468,528,472,1524,1472,528,1468,528,476,1528,1468,528,472,1524,476,1524,1472,348,1148,1092,404,1092,404,1092,408,344,1144,356,1144,348,1096,400,352,1144,352,1144,352,1144,1096,404,348,1148,348,352,1144,348,1148,348,1152,352,
15:33:25.839 -> 
15:33:25.885 -> Decimal: 2812934867 (32Bit) Binary: 10100111101010011111101011010011 Tri-State: not applicable PulseLength: 704 microseconds Protocol: 2
15:33:26.023 -> Raw data: 7048,1472,528,472,1524,1476,524,476,1524,476,1524,1472,528,1472,528,1472,528,1472,528,472,1528,1472,524,476,1524,1476,528,472,1528,472,1528,1472,1096,404,344,1152,1092,408,1088,404,348,1148,348,1148,348,348,1148,348,1148,348,1148,1092,404,348,1148,348,1148,348,352,1144,348,1152,348,1152,348,
15:33:26.339 -> 

OFF
15:33:29.449 -> Decimal: 2045667539 (32Bit) Binary: 01111001111011100110100011010011 Tri-State: not applicable PulseLength: 703 microseconds Protocol: 2
15:33:29.588 -> Raw data: 7036,476,1524,1468,532,1464,528,1468,528,1468,528,472,1524,476,1524,1472,528,1472,532,1464,528,1472,528,472,1524,1468,532,1468,528,1472,528,472,1092,404,348,1144,1092,404,348,1144,348,1148,348,1144,348,1092,404,1092,400,348,1144,348,1148,348,1144,1096,400,1092,348,1144,348,1148,348,1152,348,
15:33:29.908 -> 
15:33:29.999 -> Decimal: 370514067 (32Bit) Binary: 00010110000101011001100010010011 Tri-State: not applicable PulseLength: 703 microseconds Protocol: 2
15:33:30.135 -> Raw data: 7032,476,1520,476,1520,476,1520,1472,524,476,1520,1472,528,1468,528,472,1520,476,1524,476,1520,476,1520,1476,524,476,1524,1472,524,476,1524,1468,352,1144,348,1152,1088,400,348,1144,348,1148,348,1144,348,348,1148,1092,400,1096,400,1092,400,348,1144,352,1144,348,348,1144,348,1144,352,1152,344,
15:33:30.459 -> 

Have you found a solution yet?

chazp246, 2022-03-17:
Well i will be the third.... I maybe found the problem. Its highly possible that it is more than 32 bit system. I have not tried it yet but in https://github.com/sui77/rc-switch/pull/405 someone added 64 bit support. The library can handle 32 bit only.