sui77/rc-switch #459: Not resieving anything

petsch9, 2022-03-30:
Not recieving anything on Port 2 of Nano. I tryed whit several boards but not succeeded.
I use a nano and connect the dataPin from the ZR2-1 to the d2 of the Nano.
also i use the example sketches

Does anyone have ideas?

dvircoh, 2022-04-18:
me too
and when I use RadioHead it worts. i don't know why

ThreepE0, 2022-06-10:
Please check my new issue: https://github.com/sui77/rc-switch/issues/466

philippedc, 2022-08-13:
Hi, I simply wanted to transmit a ON/OFF flag to an alarm, but it seems much more difficult than expected...

I made the example code tests with Arduino UNO first. Then, the objective will be to use ATtiny85.

However this library does not seem to work anymore. I've checked the #466 issue, however nothing change.
2 notes:
- I also tested manchester.h, nothing is working.
- VirtualWire that is supposed to be old (and so obsolete) still works fine. But it does not support ATtiny85 :(

IDE version 1.8.13, is it the reason ?