sui77/rc-switch #463: Stale repo, PR's going nowhere?

greiginsydney, 2022-05-16:
Hi @1technophile ,

I see you have write/contributor access to this repo and released the last update back in January 2021.

Lots of people are still blindly contributing PRs but they seem to be going nowhere.

Without an active maintainer is there a point to this, or are their efforts being wasted - or is there a more "lively" fork I should be following (like your own)?

I have a vested interest: I'm using this repo for a soon-to-be-released project of my own, and I'd hope some of those PRs might be accepted to improve the reliability and range of RC devices supported...

\- Greig.

1technophile, 2022-05-18:
Hi @greiginsydney

Yes, I'm focusing on BLE nowadays but keep an eye on RF.
Which PR are you thinking on submitting?

Locke, 2022-05-19:
Well, I would not forbid new PR creation. People should see that PRs are rarely / not merged.

But creating PRs gives them the possibility to at least offer their own changes & forks to others, who might be interesting in the same change, and could then just pick that instead of writing it again and again. Therefore I'd say it not completely wasted effort to have new PRs.

greiginsydney, 2022-05-21:
Don't worry @Locke I'm not advocating that the creation of PRs be blocked, but I'd love to see more of that effort incorporated into _this_ repo by merging them rather than spawn 38 (as at today's PR count) new forks.

Is there a way of breathing life back into this repo, or am I flogging a dead horse?

greiginsydney, 2022-05-25:
Hey @1technophile , if I was to consolidate a bunch of PRs that add new remote codes (only), would you be amenable to merging the PR and optionally crafting it as a new release?

I can see 6 that are just a single line addition each.

1technophile, 2022-05-29:
Hey @greiginsydney, yes but my main concern is that I'm not able to verify that the previously compatible devices will work.
If we can have testers helping and confirming the branch I will be happy to do so.

greiginsydney, 2022-05-29:
I guess this is the problem with any large open-source project: how can you be sure the code revisions are good?

I've confirmed the addition of the now five new codes (*) don't break the ones I'm using, but that's as far as I can go with it.

(*) I e-mailed one of the authors with a query and they haven't responded.

I'll formalise the PR in the coming day(s) and we'll see what comes of it. Fingers crossed.

\- G.

greiginsydney, 2022-06-01:
Just created PR #464

mbwhite, 2022-07-03:
On a related note, I noticed that WiringPI has stopped been supported.  To the extent that the source isn't available anymore.

Alternatives are available, happy to help if I can..