sui77/rc-switch #468: Protocol Sync after sending

Ab3lqpasa, 2022-07-18:
Hi,

First of all sorry because my english is not good.
I just tried to send raw data using a protocol and I saw that the sync bit of the protocol is sent after the code instead of sending it before the code (code still works if setRepeatTransmit>1 because the sync code of one transmit is used on the next) .
Is just my protocol that needs sync bit at first?

I just changed this:

```
void RCSwitch::send(unsigned long code, unsigned int length) {
  if (this->nTransmitterPin == -1)
    return;

#if not defined( RCSwitchDisableReceiving )
  // make sure the receiver is disabled while we transmit
  int nReceiverInterrupt_backup = nReceiverInterrupt;
  if (nReceiverInterrupt_backup != -1) {
    this->disableReceive();
  }
#endif

  for (int nRepeat = 0; nRepeat < nRepeatTransmit; nRepeat++) {
    this->transmit(protocol.syncFactor);    //ADDED------------------------------------------------------
    for (int i = length-1; i >= 0; i--) {
      if (code & (1L << i))
        this->transmit(protocol.one);
      else
        this->transmit(protocol.zero);
    }
   // this->transmit(protocol.syncFactor);   //DELETED----------------------------------------------------
  }

  // Disable transmit after sending (i.e., for inverted protocols)
  digitalWrite(this->nTransmitterPin, LOW);

#if not defined( RCSwitchDisableReceiving )
  // enable receiver again if we just disabled it
  if (nReceiverInterrupt_backup != -1) {
    this->enableReceive(nReceiverInterrupt_backup);
  }
#endif
}
```

casiobearing, 2023-02-15:
I agree with this; the sync pulse should be the first thing sent in a transmission.

geoffday67, 2023-06-18:
I agree too - I changed this locally and transmission was more reliable. It seems more logical.