sui77/rc-switch #470: Can't receive data from thermo/hygrometer "TFA Dostmann 30.3249.02"

ThFischer, 2022-09-19:
Hello,
I try to receive data from a "TFA Dostmann 30.3249.02" (thermo/hygrometer).
The payload is similar to the [TFA 30.3221.02 protocol](https://github.com/RFD-FHEM/RFFHEM/blob/a1222ecd68e1febeca894ebdabfe71c22ae7fa06/FHEM/14_SD_WS.pm#L591) which I could validate with a LogicAnalyzer.
The problem is: Neither rc-switch's ReceiveDemos nor  the ProtocolAnalyzeDemo outputs any data (other transmitters are well recognized).

The signal looks like this:
[image]

1. frame-sequence of either 3 or 4 impulses ~750µs High / 700µs Low
2. payload (41 bit)
3. frame sequence of 4 impulses ~750µs High / 700µs Low
4. payload repeated
5. ...

The payload is sent 4 times, at the end no frame sequence is sent.

What really scares me:
When I tried to send only the payload with another Arduino  like:
`mySwitch.send("11110001011000101110000000101100100001001");`
I also not receive any data from ProtocolAnalyzeDemo  😱

My LogicAnalyzer schows me:
[image]

which wouldn't be the sent message, just partial similar:
```
   11110001011000101110000000101100100001001     (sent)
000000000110001011100000001011001000010010   (received)
```

Is there any chance to get this signal recognized by rc-switch?

Thanks in advance
Thomas

ThFischer, 2022-09-22:
Hi,
After figuring out that maximum is 32 bit I added support for 64 bit messages.
I added also this protocol definition:
```c
  { 245, {  3,  3 }, {  1,  2 }, {  2,  1 }, false }
```

The message is still not recognized.
If I change the protocol to have longer sync pulses it is at least recognized as protocol 6:
```c
  { 245, {  19,  3 }, {  1,  2 }, {  2,  1 }, false }
```
With less then 19 it fails again.

Does any body has an idea how to change rc-switch to work with shorter sync pulses?

Kobbe1, 2024-01-17:
Did you ever get this working? It does work for me with rtl_433 and also https://github.com/AK-Homberger/Bresser-3CH-433MHz-T-H-Sensor-decoder/tree/main

On rc switch I had some limited success with this fork: https://github.com/Martin-Laclaustra/rc-switch/tree/protocollessreceiver

But so far can't get it recognized as protocol ...

ThFischer, 2024-01-23:
Hi @Kobbe1 ,
yes, it works as expected, see my repo
https://github.com/ThFischer/rc-switch-64bit