sui77/rc-switch #478: can't decode 433mhz with esp32

xBLACKICEx, 2023-01-19:
Hi I'm trying to decode a remote control but no matter how I press it I don't get any messages. Please help. I have tested the receiver from the transmitter(SendDemo) and it is working

The sample I used is ReceiveDemo_Advanced

Some manufacturing data of my remote control
[image]
[image]
[image]

Jadyla, 2023-01-24:
Hey! Did you tested with another GPIO? If I am not mistaken, the code starts using GPIO 0.
Esp32 has some particularities, including the pinout and pins functions. It might not be the problem, but it's something to test and discard.

xBLACKICEx, 2023-01-24:
> Hey! Did you tested with another GPIO? If I am not mistaken, the code starts using GPIO 0. Esp32 has some particularities, including the pinout and pins functions. It might not be the problem, but it's something to test and discard.

The problem you mentioned I should have encountered during the test and have ruled it out. I need to use digitalPinToInterrupt() to do so, otherwise I can't receive any information.

However, the message from the remote control cannot be received.

Jadyla, 2023-01-25:
Recently I also used esp32 with RF receiver. I had some problems because I was using GPIO 22, that is also an UART, and I think that somehow this was causing some conflicts when booted. Change the GPIO solved my problem.

Another change that I made was use 3.3V, not 5V.

I did not use `digitalPinToInterrupt()` for esp32, because of this I think that maybe this is not the problem (this is used on Arduino because only some pins can support interruptions).

Did you received some warning? Or maybe some information on Serial monitor.

dvdbrk, 2023-02-23:
I have maybe the same problem, and I think I have tried everything ever mentioned as a possible solution, but still no luck. I have opened the new issue which shows my wiring as well.

NorthernMan54, 2023-02-23:
I think your issue is documented here - https://www.rflink.nl/wires.php

That receiver is effectively garbage.

One option, is the openMQTTGateway implementation of RF, and a decent receiver

saharesp, 2025-03-06:
receiver 315 Mhz