sui77/rc-switch #490: How can i modify this protocol?

Mavciii, 2023-05-02:
hey. i have  blind remote controller and i wanna send on, off, close command with my nodemcu.
The communication protocol was sent by the manufacturer as follows. But the code is 40 bits. How can I send commands with this?

[Wireless protocol(English)_Feierge.pdf](https://github.com/sui77/rc-switch/files/11370621/Wireless.protocol.English._Feierge.pdf)

Mavciii, 2023-07-31:
??

Mavciii, 2023-09-06:
the main problem is it. i cant configure this code. i dont know what can i do :)

darkbox, 2023-09-08:
Try with a cheap logic analyzer or an oscilloscope if you have access to one.
The protocol is given for what I understand by the times periods of the signal being high and low.
Sorry for not been more helpful at this point :anguished: