sui77/rc-switch #502: 433MHz 4-button code copy

bsdinis1, 2024-02-22:
I've reverse engineered a cheap 433MHz remote with 4 buttons and code copy. I'm trying to understand how it works so that I can interface it with my own microcontroller.
I'd very very grateful if anyone can explain how this works and how can I interface this circuit with my own ESP32.

The red rectangle is the [FT60E021](https://file.elecfans.com/web1/M00/EC/05/pIYBAGB-mVuAZM2zAFOHEdeP_bw832.pdf) microcontroller with all of its pins labeled.
[Q1 and Q3](https://datasheet.lcsc.com/lcsc/2112011530_JSMSEMI-2SC3356-R25_C963378.pdf) are Silicon Epitaxial Planar Transistors (marking R25).
[Q4](https://datasheet.lcsc.com/lcsc/1810161230_Jiangsu-Changjing-Electronics-Technology-Co---Ltd--S8050-J3Y_C2146.pdf) is an S8050 general purpose NPN transistor (marking J3Y).
[Q5](https://www.onsemi.com/pdf/datasheet/mmbth10lt1-d.pdf) is an MMBTH10L general purpose NPN silicon transistor (marking 3EM).
[Q6](https://www.rcscomponents.kiev.ua/datasheets/s8550-2ty-datasheet.pdf) is a general purpose PNP transistor (marking 2TY).
[X1](https://www.wtlcrystals.com/uploads/files/products/product-file-161372362452.pdf) is a SAW resonator (marking B01).

[image]

Alyxvance, 2025-03-18:
wahts not undestandable it learns the code and send ist out

bsdinis1, 2025-04-11:
> wahts not undestandable it learns the code and send ist out

great, but I already understand the device's functionality...
what I'm asking is how does this circuit work, what is the purpose of each of the transistors, which pins does the microcontroller pull-up/-down to transmit/receive, etc.

bsdinis1, 2025-04-20:
updated schematic in first post.

BiancoMarco, 2025-06-12:
... for the record:  I did the "Simple way" for my garage, but I may in the future play with the advanced way