sui77/rc-switch #508: Have anyone try Rcswitch with esp32c3 supermini

feyhong1112, 2024-04-07:
The issue that I am facing now is the recieve pin is did not work anyone else go this problem?

lllwan, 2024-05-04:
I have also encountered this problem but it remains unsolved.

feyhong1112, 2024-05-11:
Hi IIIwan, 
Sorry about late reply. I hope you still keep going on this work.
I wait about 2 week to get a new syupermini board. The reason that I want to buy a new one because I am curious is that
1) I broke the board by connect module on 5V to transmit the code
2) I suspect that the board itself have malfunction since it shiping to me <----I think this is the most true reason becuase on rf433 module, I think it should have smd transistor to seperate different voltage

I am now changing to lib "smartrc cc1101" by Lsatan with cc1101 3.3V on now new supermini esp32c3 board
What I notice is that the pin that I use to receive is pin 4 without needed to be use digitaltointerrupt or any spacial method. Oh yes, I setting upload with JTAG disable, erase all flash*...: disable and if it work on me I guess I hope that it would be work on you also. Try on different supermini borad 

If you switch to cc1101 please don't for get to change spi pin dont trust the google supermini spi pin image. Use the chinese vendor data wire.
I am also attach mycode